RosettaDrone/rosettadrone #2: Make it work on DJI Mavic AIR and other drones

The1only, 2020-03-07:
It would be nice to use Rosetta Drone 2 on all DJI drones. Today I bought a brand new Mavic AIR and will start the integration tonight.

The1only, 2020-03-07:
Amazing new, I connected the Mavic Air, that got a very different video format. However,  the DJI SDK seems to handle this, and both video and telemetry was working without any modification. This might prove that all SDK supported DJI drones will work. I will test more to morrow. And then I can clean up the code, removing all the old Video code.

BkMak16, 2020-03-27:
I'm sure this isn't the place to ask but I'd rather not start a new issue for a relatively minor question. My apologies there.

I just wanted to find out if there is a way to get a video feed or any visual data from the Mavic 2 Pro that isn't from the main camera. So to say one of the downward facing cameras?

The1only, 2020-03-28:
No unfortunattly they can only be used as sensors. You got to use the Matrice to get multipe video streams.

closed 2023-02-24 by kripper