RosettaDrone/rosettadrone #5: Cant build in android studio

AryanAfrouzi, 2020-03-27:
Could be a problem on my end but for some reason, I can't run the app on a device, the project builds but when I go to run on my device I get error: undefined reference to 'g_thread_self' from video_encoder.c file. Also here is the command at which is fails at:

Build command failed.
Error while executing process C:\Users\aryan\AppData\Local\Android\Sdk\cmake\3.6.4111459\bin\cmake.exe with arguments {--build C:\Users\aryan\Downloads\rosettadrone-4.11.1.2\rosettadrone-4.11.1.2\app\.cxx\cmake\debug\armeabi-v7a --target VideoEncoderJNI}
[1/2] Building C object CMakeFiles/VideoEncoderJNI.dir/src/main/cpp/video_encoder.c.o
[2/2] Linking C shared library ..\..\..\..\build\intermediates\cmake\debug\obj\armeabi-v7a\libVideoEncoderJNI.so
FAILED: cmd.exe /C "cd

AryanAfrouzi, 2020-03-28:
upon further review I cant even seem to find the g_thread_self function, I could be blind

The1only, 2020-03-28:
What branch are you on?

Best regard
Terje Nilsen
9Tek AS
Norway

AryanAfrouzi, 2020-03-28:
> What branch are you on? Best regard Terje Nilsen 9Tek AS Norway
> […](#)

I pulled from master and built the project after adding my dji key

The1only, 2020-03-29:
You are right, I was cleaning up dead code and was too quic. Fixing it now... have erased and do a checkout now to verify.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-03-29:
Ps:  if we can work out the Mavic Air video format, we can remove all the ffmpeg, gstreamer and more code. Make The project code way less complex.

Best regard
Terje Nilsen
9Tek AS
Norway

AryanAfrouzi, 2020-03-29:
all works now sir and I will also be willing to work out all the formating sounds like a great use of time :)

The1only, 2020-03-29:
Ok perfekt.

tips:  if you tapp 5 times on the drone symbol you can enter the software without connecting a drone. (For GUI debugging)

In the drone settupp you an enter a name that will show up when the software starts, to make it more private.

I have added an AI button that a python script running Dronekit can subscribe to, to controll the custom AI, this will be pusshed tonight.

Best regard
Terje Nilsen
9Tek AS
Norway

mrosel, 2020-04-14:
@The1only, this has regressed in master. :disappointed: 
I could only get a successful build if I went back to [March 29 commit](https://github.com/The1only/rosettadrone/commit/d0773c370f048a729e35849df9b4dc7563631290)

The1only, 2020-04-14:
Can you send me the error message? I’ll fix it. There was an issue, but it is suppose to be fixed.

Best regard
Terje Nilsen
9Tek AS
Norway

mrosel, 2020-04-14:
Looks to be the same as the original of this thread with reference to `g_thread_self`
```
rosettadrone/app/src/main/cpp/video_encoder.c:52: error: undefined reference to 'g_thread_self'
```
This PR fixes it for me - https://github.com/The1only/rosettadrone/pull/9

The1only, 2020-04-15:
That problem was fixed some time ago. I did a clean checkout of main now and it compiles and downloads just fine. Try to make a new directory and then checkout again to be sure you get the latest.

mrosel, 2020-04-15:
It compiles for you? Strange, starting from a fresh folder/git pull, I added back those c-libs until the compile errors went away. Turned out the whole directory `app/src/main/cpp/lib/` was needed.
Is it possible that you have these libraries linked without having them explicitly in the repo?

The1only, 2020-04-15:
No, but I always use git clone --recursive ( is it the recursive that saves me???)

The1only, 2020-04-15:
Hmm very strange, if you browse the source tree on the net (in the web browser) all files are there and no linking...

mrosel, 2020-04-15:
I tried that too. Same error. 
Steps to reproduce:

- Checked out a fresh copy (with git clone --recursive) on a Fedora workstation, 
- Removed `.idea/`
- booted Android Studio 3.6.2 (very recent). 
- loaded project. file->new->import project
- added my dji sdk key
- run build-> make project

`error: undefined reference to 'g_thread_self'`

I do see this error but not sure if it's related
```
10:54	Unsupported Modules Detected: Compilation is not supported for following modules: 9TekAS-work_9_all_teni-rosettadrone, dji-Linux-rosettadrone, Linux-rosettadrone, 9TekAS-work_9_all_teni-rosettadrone-app, work_9_all_teni-rosettadrone, rosettadrone-app, rosettadrone, Linux-rosettadrone-app, Documents-9TekAS-work_9_all_teni-rosettadrone~1, Documents-9TekAS-work_9_all_teni-rosettadrone, work_9_all_teni-rosettadrone-app. Unfortunately you can't have non-Gradle Java modules and Android-Gradle modules in one project.
```

mrosel, 2020-04-15:
I do see in [this commit](https://github.com/The1only/rosettadrone/commit/a1769bf27969d1ceb7bbc21323fbfe6733c01496) where you removed all the files in `app/src/main/cpp/lib`. That gave me the idea to add them back, which worked. ATM, I can't think why it would be different between you/me (Ubuntu/Fedora).

The1only, 2020-04-15:
Thank you, this is my mistake, I must have done something wrong when adding back the files, I am starting to belive I added some files that should not have been added. Especially since you can se my old dead built, that I do not know how to remove.

The1only, 2020-04-15:
1) The .idea should probably not be checked in ????

The1only, 2020-04-15:
PS: So you got the 3.6.2 working ??? I am still at 3.5.3 as upgrading gave me several issues...

mrosel, 2020-04-15:
> 1. The .idea should probably not be checked in ????
I'm not sure the best practice here. If I don't delete the `.idea/` folder, I get
```
10:54	Unsupported Modules Detected: Compilation is not supported for following modules: 9TekAS-work_9_all_teni-rosettadrone, dji-Linux-rosettadrone, Linux-rosettadrone, 9TekAS-work_9_all_teni-rosettadrone-app, work_9_all_teni-rosettadrone, rosettadrone-app, rosettadrone, Linux-rosettadrone-app, Documents-9TekAS-work_9_all_teni-rosettadrone~1, Documents-9TekAS-work_9_all_teni-rosettadrone, work_9_all_teni-rosettadrone-app. Unfortunately you can't have non-Gradle Java modules and Android-Gradle modules in one project.
```
I 'think' the later jetbrains fixed needing this. I'll need to research that more.

The1only, 2020-04-15:
I updated to 3.6.2, it did not work at all, but then I found that I now could upgrade Gradle to 6.3 and then I was able to build. So now I am also on all the latest.

Again I appreciate you're time.

mrosel, 2020-04-15:
> PS: So you got the 3.6.2 working ??? I am still at 3.5.3 as upgrading gave me several issues...

Yep!
```
Android Studio 3.6.2
Build #AI-192.7142.36.36.6308749, built on March 18, 2020
Runtime version: 1.8.0_212-release-1586-b4-5784211 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Linux 5.5.10-100.fc30.x86_64
GC: ParNew, ConcurrentMarkSweep
Memory: 1237M
Cores: 12
Registry: ide.new.welcome.screen.force=true
Non-Bundled Plugins: 
```

The1only, 2020-04-15:
But now the code does not run any more thow... When I install it, it just locked up. Must investigate. Can you actually run the code on an android device?

The1only, 2020-04-15:
I am back to Android Gradle Plugin 3.5.3 and 6.2.2 Gradle, and everything works again, guess that update needs a closer inspection...

mrosel, 2020-04-15:
> But now the code does not run any more thow... When I install it, it just locked up. Must investigate. Can you actually run the code on an android device?

It crashed in my emulator (but I ran it on x86). I put the apk on two devices and it worked as expected. Not well tested, yet...  I am planning to hook it up to my Air some time today and test it that way.

in `app.gradle` I did set 
```
            abiFilters 'armeabi-v7a', 'x86', 'arm64-v8a'
```

Ran cleanly on a modded Nexus 5, arm64, Android 9.

The1only, 2020-04-15:
PS: Are you working on your FORK or my main tree???? Maybe you forked when the bug was there???

The1only, 2020-04-15:
Yes it crashes on the emulator unfortunately, it always did. Do not know if this is caused by the DJI modules or something else... Would have been nice if it did not.

The1only, 2020-04-15:
I accepted your PR, maybe it relinked something that would help. 
I se no difference.

kripper, 2023-04-08:
Please check latest version.

closed 2023-04-08 by kripper