RosettaDrone/rosettadrone #7: No GPS

d4nidroid, 2020-04-01:
Hello, are you receiving GPS on qgroundcontrol?

The1only, 2020-04-01:
Yes….  Is there a problem?

droneslife, 2020-04-01:
yes I see 0 GPS

The1only, 2020-04-01:
Got 17 satellites, and I got velocity and position...
This is Mavic 2 Pro

I only uses the QGC 3.5.6 as the 4.x got crappy video…

The1only, 2020-04-01:
You are outside right? Sorry just had to ask…

droneslife, 2020-04-01:
yes I can see 16 sat on the cellphone but on the QGC no sat so maybe is the version I will change because video also crappy grey

The1only, 2020-04-01:
Do you se 16 satellites in the Rosettadrone 2 software, ok 

And you got the right IP and PORT and the system says connected??

And you got a Mavic 2 Pro ?

The1only, 2020-04-01:
Yea , the gray video is a problem. On the 3.5.6 the video got no gray problems and it has almost no latency.
There is something we need to understand there, as I get gray with my gStreamer also…

The1only, 2020-04-01:
PS: I am in contact with the QGC guy’s to fix the video issue.

droneslife, 2020-04-01:
im not on 3.5.6 and same grey color on the video

droneslife, 2020-04-01:
oh ok after a while it looks better just some minutes

The1only, 2020-04-01:
You got video issues on 3.5.6 ???

droneslife, 2020-04-01:
what do you use on video bitrate and video encoding speed?

The1only, 2020-04-02:
I do not transkode, just forward from DJI, the speed you can sett in the Rosettadrone app.
Might be I should transcode using gStreamer. The code is there, I just dont whant the added delay.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-04-02:
They say we should use QGC 4.05, for the video issue, it’s a bug and they will backport.

Best regard
Terje Nilsen
9Tek AS
Norway

droneslife, 2020-04-04:
ok! thx I will try again, Sr what is the file on the project where you add more flight modes and joystick?

The1only, 2020-04-04:
DroneModel is the file that does all that. Together with MavlinkReceiver ...

Let me know if you make something we all may need.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-04-04:
PS: Joystick you add on the QGC, you can then fly with a Joystick connected to the PC, this is already implemented. Also reading the DJI controller stick on the QGC/Dronekit  is implemented.

droneslife, 2020-04-08:
Thanks!! other question is the geo fence working?

The1only, 2020-04-08:
Not tested, but is should.

Best regard
Terje Nilsen
9Tek AS
Norway

kripper, 2023-03-30:
Sometimes it's hard to init the GPS. Not related to Rosetta.

closed 2023-03-30 by kripper