RosettaDrone/rosettadrone #8: Making Video work on Android 5.1

MrTaquion, 2020-04-04:
I was was able to compile (master branch) and run Rosetta Drone 2 in a Android 5.1 phone (i do not have an Android that supports 8.0). To do this I changed the "targetSdkVersion" and "minSdkVersion" to 22.

The first minor issue I had was that the LinearLayout showing the PreFlightStatusWidget, FlightModeWidget, etc. They did not enter the screen fully, leaving the settings and other important buttons out of it. I simply changed the order to make the settings button accessible. This was an issue only on the phone (LG k4 -- 5'' of screen) and not on the android emulator using a phone with 6'' of screen size.

I was able with this setup to obtain telemetry data both in QGroundControl and MAVProxy, however I did not get Video feed. The DJI_LOG did not show any errors, aside the message: "Transcode Video !!!!!!!". 

I was able to get video feed by doing the following modifications in "MainActivity.java":

Replaced line 581
`mGstEnabled = false;`
with
`mGstEnabled = true;`

Replaced line 219
`if (!isTranscodedVideoFeedNeeded()) {`
with
`if (isTranscodedVideoFeedNeeded()) {`

However I kept getting often a "Message too long" for the sent DatagramPacket in line 552, which resulted in a poorly transmitted video stream (lots of missing frames). I associated this with the slow processing capabilities of the phone so I tried to remove the video preview on the phone by modifying line 245 (to INVISIBLE) which broke the video stream. Instead I made the video preview smaller (150dp x 75dp) in "activity_gui.xml" which resulted in a good and steady video transmission (no message too long errors) which I tested using ffplay and ffmpeg on the incomming port "udp://X.X.X.X:5600".

I do not fully understand why this fixed the video issue, since I do not understand the meaning of "TranscodedVideoFeedNeeded" or the need to sometimes enable "mGstSocket" and sometimes not. 

I would appreciate any comments on this, specially on why these modifications result in the video working. Any comments to increase performance are welcomed too (lower the fps in the transmitted video, etc).

The1only, 2020-04-04:
Greate!!! I never had the video issue, but then I use newer faster phones and tablets.

The too long issue probably is caused by not running the Nails part... I also had that problem. I will look at you’re suggestions asap.

But most importantly what DJI drone do you use?  Mavic 2 Pro????

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-04-04:
I now understand what happens.

You're system is to slow to run the Nails ( making nice h264 UDP frames, a lot of java code) so you just forward what comes from the DJI directly.
When I did that I got the large frames all the time.
Then you make the image smaller ( it must be there is’s a DJI thing)  and then the system was fast enough. 

I am not sure how to solve this, as we really should run the Nails …
The reason for the optional path ( that you now enabled) it that I have issues with Mavic Air, it uses a different format that I can not decode. 

If we can measure the phone performance we might be able to adapt.
However for now I am glad you discovered this, but I belive we should keep the main code as is.

The Transcoding is something DJI added fro Mavic 2 series and newer systems, they add some additional data to the frames, and we need to run ti through the transcoder to get H264, however the older drones does not need this. 
You're changes did not alter the Transcoder only skipping the nails...

Terje

MrTaquion, 2020-04-04:
Thanks for the quick reply!

> But most importantly what DJI drone do you use?  Mavic 2 Pro????
Yes, I'm using the Mavic 2 Pro. Still haven't tested if MAVLink plans work but I expect they do.

> If we can measure the phone performance we might be able to adapt.
> However for now I am glad you discovered this, but I belive we should keep the main code as is.
Of course, besides of wanting to understand the fix I wanted to list the issue in case anyone else tries to run the app on low end hardware. Thanks a lot for the insight! I'll see if I can manage to run the Nails part because I still get some large UDP packages just after startup.

As a side note, I'll be looking into implementing on the app other functions from the DJI API, as the intelligent modes (ActiveTrack specially) and such. Are they already implemented on the app? Or should I start from scratch (and where do you suggest for this code to live in)?

The1only, 2020-04-04:
If you got the latest code, the MAP and mission is implemented ( enter it from the settings, not tested and I still got some issues with Google map, remember you need you're googlemap license key also then, free thaw) but other that that it's not there ( preflight list is there hitt the connection sign). However is all in the API and all you need to do is define it in the XML file and DJO does the rest for you (almost) And you find all shows I already implemented.  Activity_gui.xml

The1only, 2020-04-04:
I got a video showing I got amazing low latency now on Ubuntu 18.04, I almost does not want to reboot my PC. I am down to one frame latency from the DJI video on the phone to the gStreamer image on my PC, 1 frame is 1/20 sec... so the only way that is even possible is that Rosettadrone 2 sends out the h264 before or while it is decoding it, so gStreamer on a laptop can catch up... I'll try to get the video out....  It tok me some 3 days to compile all the tools who NVIDIA support etc...

The1only, 2020-04-04:
https://www.youtube.com/watch?v=HAC5OpQQpDI&feature=youtu.be

The1only, 2020-04-04:
PS: Be aware that you can tap 5 times on the drone image at the boot screen to bypass the need for connecting a drone ( for GUI work)

Also you can tap the simulator to test you're flying without actually fly, but then you need the Drone connected.

MrTaquion, 2020-04-04:
Saw the video, amazingly low latency! I was able to achieve with my setup around 1 or 2 seconds delay.

> PS: Be aware that you can tap 5 times on the drone image at the boot screen to bypass the need for connecting a drone ( for GUI work)

Yes, I saw this feature, it's very handy.

> Also you can tap the simulator to test you're flying without actually fly, but then you need the Drone connected.

Haven't figured this one out yet, is the simulator a button in the GUI?

The1only, 2020-04-04:
On the start screen, just tapp it one time and you will get it. Then takeoff and you’ll se 👍

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-04-04:
The simulator needs the drone connected thow... DJI thing...

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-04-04:
Ps: you can enter you’re name in the drone settings menu, to make the startup screen more persional. Just a small thing to impress customers and friends...

Best regard
Terje Nilsen
9Tek AS
Norway

MrTaquion, 2020-04-04:
Thanks a lot for all this info!!
Cheers!

BkMak16, 2020-06-22:
Do you have the steps how you achieved such low latency on video feed. 

Having a grey stream with maybe 3 seconds per frame, at best 1.5 seconds per frame while Streaming to QGroundControl. Seconds, per, frame. 

Not sure if I need gstreamer because I'm having trouble there aswell. Need to know if it's necessary to put the work into learning gstreamer to and fix it and get the frames.

The1only, 2020-06-22:
I used QGC 3.5 something, but also gStreamer in Python. QGC 4.x got some issues and are worthless. Then set the low-latency flag, make sure the system uses GPU, I use OpenCV that I compile myselfe,  and tweek everything. Gray frames are normally caused by to slow performance of the PC system. I use Ubuntu 18.04 with Nvidia 1650 video/GPU card. 

I belive the gstreamer command line (several lines) are in the code someware, I’ll check.
I have one gStreamer commandline for sending and one for receiveng (for simpler test) then when it works I use the Drone for final test.

PS: Connected a Matrice 210 V2 the other day, worked out of the box, but with some graphics positioning issues on the crystal Sky only...

Best regard
Terje Nilsen
9Tek AS
Norway

BkMak16, 2020-06-22:
Awesome. Will give it a shot!

BkMak16, 2020-07-04:
Couldn't get the stream working. 

Had to find a work around but managed.

Thanks!

kripper, 2023-04-08:
See: https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/edit#heading=h.q9mp6yjsfxe5

closed 2023-04-08 by kripper