RosettaDrone/rosettadrone #10: Clarifying the RTP streaming quality settings

mangolas, 2020-05-07:
Thanks for making this, this is great resource and I like the idea of being able to use MAVLink on top of DJI SDK.

I managed to get Rosettadrone running on Smart Controller and Mavic 2 Enterprise Dual with no problems, but I would like to understand how to get more than 640x480 RTP output. I'm quite novice on video streaming matters, so I might just not understand this right, but nevertheless I have tried setting `videoBitRate` to different levels (tried 3, 4, 10), but when I play RTP stream, it's always 640x480. I could not find another setting that I could use to affect the outgoing video stream.

The GStreamer command I've been using to show incoming feed is `gst-launch-1.0 -v udpsrc port=5600 ! application/x-rtp, encoding-name=H264 ! queue ! rtph264depay ! h264parse ! avdec_h264 ! autovideosink`.

So how one could control the outgoing video size? Or is it just wrong GStreamer pipeline config?

The1only, 2020-05-08:
Interesting, I belive I always get 1280*720... meybe I am wrong, I will check. I do forward what I get from DJI.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-05-08:
> 8. mai 2020 kl. 05:05 skrev Terje Nilsen <teni@9tek.no>:

When configuring the drone with Go4 how did you set this setting? 

Best regard
Terje Nilsen
9Tek AS
Norway

mangolas, 2020-05-08:
I think I realised the reason for 640x480, I had thermal camera on which only have that resolution. 

Now that I'm using normal camera with Rosettadrone, GStreamer is reporting getting 960x720, which is better, but still not 1280x720 I was expecting. Maybe that is then Mavic 2 Enterprise Dual native h264 resolution, but I'm not quite sure if there is any way to control that resolution.

The1only, 2020-05-08:
Ok, I have only teste Mavic 2 Pro, it only got one camera and two live view modes HD or FullHD.

Best regard
Terje Nilsen
9Tek AS
Norway

kripper, 2023-04-08:
See https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/edit#heading=h.q9mp6yjsfxe5

closed 2023-04-08 by kripper