RosettaDrone/rosettadrone #12: No GCS connectivity

tareed, 2020-06-24:
**Describe the bug**
Using both a Phantom 3 Standard and a Matrice with Rosseta Drone on a Samsung tablet we are getting spotty connection to a GCS. We have tried our own GCS and QGroundControl.  It will work then not work for a while even after multiple reboots of all systems. Basic network connectivity checks out. 

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on Drone
2. Connect an tablet to the WiFi 
3. Setup the IP for the remote GCS in Rosetta Drone
4. See that no data makes it to the GCS 

**Expected Behavior**
Connection to GCS works constantly 

**Screenshot**
None

**Desktop (please complete the following information):**
 - OS: Phantom 3, Matrice, Android 9.0
 - Browser 
 - Version 

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

The1only, 2020-07-04:
Newer had that issue, you might get bad video normally caused by performance lacks on the desktop PC. 
But the telemetry and video ALWAYS show up on the QGC screen. 
However, if you change settings you might need to restart Rosettadrone (it should remember all settings), and remember to do this you MUST use Android settings and force it to stop, or it will stay resident. 

This is NOT a general issue, so there must be something with the setup.

tareed, 2020-07-04:
I will retest with your recommendation but this persisted even after reboots of the gcs, drone, tablet and controller.

kripper, 2023-03-11:
Probably fixed. Please retest.

closed 2023-03-11 by kripper