RosettaDrone/rosettadrone #14: Pan and Tilt of camera gimbal

BkMak16, 2020-07-04:
The Pan and Tilt of camera gimbal works fine as expected on the ground with the def setServo.
However in flight, the servo 8 which pan's the gimbal, now yaw's the craft.

Land again, this works perfectly,

So now the craft doesn't pan the camera but does tilt the camera, 

If you yaw normally the craft yaw's twice, either with the setServo direction or against the direction.
Tried using the Mav ROI instead, that doesn't move the gimbal at all.

If the gimbal pan was set to an awkward angle on take off, this cannot be corrected in flight, even though the tilt can.

Any ideas on how to fix this?

The1only, 2020-07-04:
Thank you, this is a new one I was not aware of. It must be a DJI sdk thing. I too will need this so I’ll investigate.

If any other have any idea’s please share.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-07-04:
Seems the gimbal must be inn YawFollow mode, and not in FREE mode. 
It is suppose to change mode automatically.

Try changing the code like this and test:

   /********************************************
     * Motion implementation                    *
     ********************************************/

    void do_set_Gimbal(float channel, float value)
    {
        Rotation.Builder builder = new Rotation.Builder().mode(RotationMode.ABSOLUTE_ANGLE).time(2);
        float param = (value-(float)1500.0)/(float)5.5;
        if (channel == 9) {
            builder.pitch(param);
        } else if (channel == 8) {
            builder.yaw(param);
        }
        if (mGimbal == null) {
            return;
        }

        **mGimbal.setMode(GimbalMode.FREE,djiError -> {
            if (djiError != null)
                parent.logMessageDJI("Error: " + djiError.toString());
        });**

        mGimbal.rotate(builder.build(), djiError -> {
            if (djiError != null)
                parent.logMessageDJI("Error: " + djiError.toString());
        });
        util.safeSleep(500);
    }

The1only, 2020-07-04:
It might not be that easy:

FREE | The gimbal can move independently of the aircraft's yaw. In this mode, even if the aircraft yaw changes, the camera will continue pointing in the same world direction. This mode is only available for the Ronin-MX when the M600 or M600 Pro landing gear is retracted.
-- | --
FPV | The gimbal's work mode is FPV mode. In this mode, the gimbal yaw will follow the aircraft's heading, and the gimbal roll will follow the RC's roll channel value. The pitch will be available to move. This mode is only available for the Ronin-MX when the M600 landing gear is retracted.
YAW_FOLLOW | The gimbal's work mode is such that it will follow the yaw. In this mode, the gimbal yaw will be fixed, while pitch and roll will be available to move.
UNKNOWN | The gimbal's work mode is unknown.

BkMak16, 2020-07-04:
Will do new builds and see what works best.

Thanks

BkMak16, 2020-07-08:
Okay,

Last question then I'll be able to give you my findings.

Is there a way to change the modes (sports, p-mode and tripod) without using the slider on the controller.  An app button, setting or line of code?

BkMak16, 2020-07-08:
On the first bootup:

_P-Mode or Tripod:_
Collision Detection		Front and Rear Only
Movement (X,Y,Z)		Yes
Yaw				        No
Gimbal 				Tilt and Pan
User Controller Input	No

_Sport:_ 
Collision Detection		No
Movement (X,Y,Z)		Yes
Yaw				        No
Gimbal 				Tilt and Pan
User Controller Input	No

After Switching Modes:

_P-Mode:_
Collision Detection		Front and Rear Only
Movement (X,Y,Z)		No
Yaw				        No
Gimbal 				Tilt 
                                        Pan makes drone Yaw
User Controller Input	Yes

_Tripod:_
Collision Detection		Full
Movement (X,Y,Z)		No
Yaw				        No
Gimbal 				Tilt 
                                        Pan makes drone Yaw
User Controller Input	Yes

_Sports:_
Collision Detection		No
Movement (X,Y,Z)		No
Yaw				        No
Gimbal 				Tilt 
                                        Pan makes drone Yaw
User Controller Input	Yes

On all modes and first boot, Gimbal Pan works properly only on the ground.

There is no way to get movement and full collision detection. To get full detection you must switch modes even if booted in tripod as you only get front and rear, switching modes causes the loss of movement and gimbal pan. To regain these, the drone must be rebooted. 

Yaw does not respond in any mode regardless of gimbal build. A form of Yaw is after a mode switch the pan becomes the Yaw, leaving no way on panning. 

A way to solve the everything except for the Yaw, is to do the system check that happens when switching to tripod which activates the remaining sensors, if the controller is booted/connected already in tripod mode so that you don’t have to switch out of tripod and back in to get the sensors, thereby losing movement.

BkMak16, 2020-07-08:
[Rosseta.docx](https://github.com/The1only/rosettadrone/files/4889499/Rosseta.docx)

Heres the WordDoc version

The1only, 2020-08-25:
My workaround is to rotate the entire drone, and only use pitch on the camera controller. One can move in any direction while pointing the head in another.

The1only, 2020-11-15:
I have investigayed a bit more, on Inspire this does not happen, seems I can not get around it on the Mavics ...

lukelu0520, 2021-01-26:
Dear rosettadrone friends and @The1only ,

I was wondering if someone tried DroneKit's Gimbal control function. We are trying to use DroneKit to change the pitch of gimbal, but not very sure if it is compatible with rosettadrone.

-- Luke

The1only, 2021-01-26:
Yes this is a big problem. It all depends on what DJI drone you use, the Inspire and Matrice shuld work as expected, however the Mavic has this issue, and if you read the DJI SDK manual there seems to be no way around it. Any suggestions are welcome. I too got issues with camera not pointing forward and it can not be adjusted in air.

I’ll fix it as soon as someone works out why!

BkMak16, 2021-01-26:
Hopefully soon I'll be able to get back to working on the Mavic.