RosettaDrone/rosettadrone #15: Mavic Air 2 Compatability?

hansonxyz, 2020-07-12:
Hello,

First, thank you for helping to maintain this project, as a fellow developer I know how time consuming these projects are.

Am I right to assume this is not compatible with the Mavic Air 2?  I built the project, it auths success to the SDK, but when I try to use it I get Status: No Product Connected.  I wonder if maybe it is not compatible with Dji Fly app?

The1only, 2020-07-12:
Hmmm no, I have tested Mavic Air 2. There was some issues with the video but not telemetry?

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-07-12:
You did turn on the drone? Sorry to ask.

Best regard
Terje Nilsen
9Tek AS
Norway

hansonxyz, 2020-07-12:
No, ask away!  I spent like 4 hours trying to get this to work yesterday,
and had to put it away.

The drone was on, I had a connection within DJI Fly.  The two things I
didn't do exactly to instruction was I put the DJI API key right in
manifest.xml (I didn't see any other references to it) and I had the drone
connected through DJI FIy (I'm unclear on if rosetta drone communicates to
DJI fly, or rosetta drone directly).

If the app is supposed to work with the Air 2, I can just start the app in
debug mode and start diagnosing it within Android Studio.  I'd also be
interested in trying to decipher the video feed stuff.  I have a major
personal project I have to get done by the end of July though, so I
couldn't contribute anything until August.

The1only, 2020-07-12:
Ahh well one thing, you must kill all DJI apps. Only one app can connect to the drone at a time. Use the built inn android app to kill the DJI apps. Lett Rosettadrone be the default app.

The API key is supose to be added as explained. 

Best regard
Terje Nilsen
9Tek AS
Norway

hansonxyz, 2020-07-12:
Thank you, I'll try this out!

lukelu0520, 2020-11-12:
I thought DJI Mobile SDK is currently not supporting Mavic Air 2 until the next release ~29th Nov. 2020.  It's kind of amazing you are able to adapt it earlier.

ref: https://developer.dji.com/user/roadmap/#msdk

The1only, 2020-12-17:
The new SDK that came out last week supports MAVIC AIR 2 officially, however, we got some issues with it, and it is only in the beta version.

kripper, 2023-02-24:
Please check latest version in master.

closed 2023-02-24 by kripper