RosettaDrone/rosettadrone #16: When running Rosetta Drone from Mavlink, and then using the remote controller to switch to a different mode, the software looses connection.

The1only, 2020-07-28:
When flying from Mavlink, using Dronekit etc. and then the operator tries to take over using the remote controller sticks, nothin happens, when the user then changes the mode to be able to do a manual takeover, the Rosetta drone must be restarted to be able to control the drone again. Seems to be a mone/connection issue...

The1only, 2020-07-28:
Normally the drone is in GPS ATTI mode when using the Mavlink the drone enters JOYSTICK mode, and the RC joysticks are disabled when the mode switch is operated the drone jumps back to GPS ATTI mode, but now never reenters JOYSTICK mode !!!!!

BkMak16, 2020-08-01:
Yes. You'll have to restart the drone and controller to get into joystick mode again.

Joystick mode only happens on the first run.

The1only, 2020-08-01:
It’s fixed just not checked inn yet

Best regard
Terje Nilsen
9Tek AS
Norway

BkMak16, 2020-08-04:
Nice one!

The1only, 2020-08-04:
I am working with absolute positioning and missions now, will putt all when it works.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-08-25:
This should now be fixed in the latest master code.

BkMak16, 2020-08-25:
Awesome!

closed 2023-02-27 by kripper