RosettaDrone/rosettadrone #17: Garbled Video in Rosetta app with DJI Phantom 4 V2 Pro and Pixel 2. Also no video to QGroundControl.

tlalexander, 2020-08-07:
**Describe the bug**

Hello!
I just compiled the app and loaded it on to a new Google Pixel 2. I have a telemetry connection and everything looks connected, but the video display is garbled. What is interesting is that sometimes I can see the image, but it is mostly garbled. See screenshot below.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile the latest release.
2. Load on to a google pixel 2
3. connect to aircraft
4. See error

**Expected behavior**
I expect to see video from the camera.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: Google Pixel 2
 - OS: Android 10

**Additional context**
I am able to send telemetry to QGroundControl but it just says "Waiting for video." Also on QGroundControl I can pick UDP h264 or UDP h265 for the video stream. Which is correct?

Thank you!

The1only, 2020-08-07:
Ahh never tested Phantom 4 V2, if it uses the old video format, like the dji Air, some people have made a trick in the video transkoder function. For the new systems like Mavic 2 Pro and Martice 210 V2 there is a new video format, this works. 

Unfortunatly I have no Phantoms so I got to put thrust in you guys. The problem is with with the transkoding, and nails.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-11-14:
Try adding Pha tom4 to the list og  of None Ocusync2 in MainActivity. Line 721 if I remember right...

tlalexander, 2020-11-18:
Oh interesting. I have moved on from DJI to an open source drone, but if I dust this off again I will try it. Thank you!

kripper, 2023-02-24:
Added

closed 2023-02-24 by kripper