RosettaDrone/rosettadrone #18: Are you already check last sdk and support for mavic mini ?

virtualrobotix, 2020-08-09:
Could nice that rosetta 2 can support mavic mini with last revision of sdk.
I havee one and i can support in test and implementation.
Some guys already test it ?
Best
Roberto

The1only, 2020-08-09:
I’ll se if I can implement the latest SDK this week. I am always a bit afraid for what DJI have changed / removed, so I might bransh it out at first.

The latest SmartController fw misses USB RNDIS ant that do create issues for our software, wlan is slower.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-08-10:
I tried to upgrade the SDK, but there are issues and the code will not run, there are also several updates needed on Android libs, and Gradel... That all creates issues...  I am spending my time on getting goto position and mission-plan to work. 
We need more people !!!

virtualrobotix, 2020-08-17:
If you tell me how i can help to support the project i can try , do you have some update ? Are your work share on git ? I find last update at 4 jul .

The1only, 2020-08-29:
The project is now updated to SDK4.13 and working. This should support Mavic MINI, however NOT tested...

virtualrobotix, 2020-09-01:
Great news i done some test today and update you.

The1only, 2020-09-01:
Seems Mini does Not use OcuSync, that might be a problem for video. I am currently working on the compatibility issu with the non OcuSync systems. I expect a beta fix within a week.

lukelu0520, 2020-11-12:
@The1only thanks for your effort on mavic mini. We tested on mini today but its video stream doesn't show up. There is message says "Image Transmission Signal Weak". We were wondering if you've updated the compatibility issue with the non-OcuSync systems. If so, is it in master or VIDEO_WORK branch ?

If you are too busy to work on it, hope you could point us a direction to work on the issue. Maybe we could contribute on the issue.

Thanks,
-- Luke

The1only, 2020-11-14:
In the MainActivity is a list of drones that needs the old handling. Try adding non working drones and retest.

avncalst, 2021-04-02:
I compiled the master branch yesterday and did a 1st test on mavic mini. First test looks promising. Congrats.

avncalst, 2021-04-08:
Weird observation: using gstreamer code to receive the video stream from my mavic mini is unsuccessful. Most of the time gray screens due to unsuccessful decoding by my old Ubuntu pc. However when I use the video tab on my QGC station (v4.0.6) I get a perfect reception of the video stream with low latency. Any ideas?
[image]

The1only, 2021-04-08:
Software performance issue.
You need gStreamer with GPU support, and very efficient code.

avncalst, 2021-04-23:
I am currently testing my mavic mini using python code. Most of the features work, however the mini does not respond to the mavlink command "send_local_ned_velocity(vx, vy, vz)". Please advise. Thanks in advance.

avncalst, 2021-04-26:
Please find hereafter some of the mavlink messages captured by logcat in response of python dronekit commands:

MOVING COMMAND + RESPONSE: mini does not really move but drifts somewhat.
MAVLINK_MSG_ID_SET_POSITION_TARGET_LOCAL_NED - sysid:255 compid:0 time_boot_ms:0 x:0.0 y:0.0 z:0.0 vx:0.0 vy:-2.0 vz:0.0 afx:0.0 afy:0.0 afz:0.0 yaw:0.0 yaw_rate:0.0 type_mask:4039 target_system:1 target_component:0 coordinate_frame:9
Message: MAVLINK_MSG_ID_SET_POSITION_TARGET_LOCAL_NED - sysid:255 compid:0 time_boot_ms:0 x:0.0 y:0.0 z:0.0 vx:0.0 vy:-2.0 vz:0.0 afx:0.0 afy:0.0 afz:0.0 yaw:0.0 yaw_rate:0.0 type_mask:4039 target_system:1 target_component:0 coordinate_frame:9

LANDING COMMAND: works correctly.
MAVLINK_MSG_ID_SET_MODE - sysid:255 compid:0 custom_mode:9 target_system:1 base_mode:1
Message: MAVLINK_MSG_ID_SET_MODE - sysid:255 compid:0 custom_mode:9 target_system:1 base_mode:1
MAVLINK_MSG_ID_SET_MODE: 9

avncalst, 2021-05-01:
Issue solved: "send_local_ned_velocity(vx, vy, vz)" does not work. This message is interpreted as a set position -> the message to be sent is (different type_mask and position coordinates used) : 

def goto_position_target_local_ned(north, east, down):
    """
    Send SET_POSITION_TARGET_LOCAL_NED command to request the vehicle fly to a specified
    location in the North, East, Down frame.
    """
    msg = vehicle.message_factory.set_position_target_local_ned_encode(
        0,       # time_boot_ms (not used)
        0, 0,    # target system, target component
        mavutil.mavlink.MAV_FRAME_LOCAL_NED, # frame
        0b0000111111111000, # type_mask (only positions enabled)
        north, east, down,
        0, 0, 0, # x, y, z velocity in m/s  (not used)
        0, 0, 0, # x, y, z acceleration (not supported yet, ignored in GCS_Mavlink)
        0, 0)    # yaw, yaw_rate (not supported yet, ignored in GCS_Mavlink)
    # send command to vehicle
    vehicle.send_mavlink(msg)

kripper, 2023-01-09:
> Software performance issue. You need gStreamer with GPU support, and very efficient code.

Sounds like a hack. Did you have problems getting the video stream via the SDK?

closed 2023-02-22 by kripper