RosettaDrone/rosettadrone #19: We need more developers involved, the project is growing out of a one man show...

The1only, 2020-08-10:
We need people to work on the upgrade to the latest SDK, I ended up with no video at all. 

We need people to work on the upgrading Gradel, it makes the code fail....

We need people to work on the Android lib uppdates.

We need more people on the Mavlink interface, there is  loot of work to be done implementing one and one feature. 

We need people on the GoogleMAP flight planner page, it does not work.

We need people to test on all different DJI platforms.

We need heads up when new things are coming our way, as DJI changes things without telling..,.

BkMak16, 2020-08-11:
I'd like to help where I can.
For now I'm busy trying to cure a pandemic, As you can imagine what its like to have the answers but lack the resources to prove it. Annoying to say the least.

When all this is over I'd love to do the testing, I use python so mavlink I think is about as far as my programming skills will help us. If resources allow I'll collect more drones and we'll get to test them out.

The1only, 2020-08-11:
Keep up the good work, I hold the forth until the pandemic has passed. The pandemic is on it's way back for a second round around here.

chron0, 2020-08-27:
I can offer testing with a DJI Mavic Pro and QGC on Linux. I've been looking forward to be able to control the Mavic outside of the DJI ecosphere for a long time, especially mission planning etc. Will try to build latest master again and see how far I can get.

The1only, 2020-08-27:
Perfect let me know. I belive I know what will fail, and can point you in the right direction. It’s the transcoding again.

Best regard
Terje Nilsen
9Tek AS
Norway

danammeansbear, 2020-10-01:
hello i am on the project. I have a mavic air, how can i test/help?

The1only, 2020-10-01:
Let me check in the latest code that will support Mavic Air.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-10-01:
> Okay I am on standby
>
> Sent from Mail for Windows 10

You can start by checking out the main code, and se if you get live video in the app and in QGC... if you do we are flying.

If you get issues just ask.

Best regard
Terje Nilsen
9Tek AS
Norway

danammeansbear, 2020-10-02:
Okay will do!

avr68, 2020-11-04:
hi
I can work on the Mavlink interface

lukelu0520, 2020-11-11:
@The1only The project is great, would help many users. Feel free to count me in. I am testing on mavic 2 zoom and mini with QGC on Ubuntu and Google Pixel 2/3/4; and will work on interfacing MAVROS and web applications.

That would be great to get an initial guide from you, so we could understand the code structure well. (even a simple flow diagram is helpful)
For example, after main/java/sq.rogue.rosettadrone/MainActivity.java, which one is the next important .java file to look at? 
I would think sq.rogue.rosettadrone/video contains the core to handle video streaming, and com.MAVLink contains the core to do the translation between MAVLink and DJI language. But maybe I miss the global picture of the code.

Best,
-- Luke

The1only, 2020-11-11:
Let’s se:

Everything that is related to Drone communication and control, except for Video happens in “DroneModel”, so any new actions is suppose to be there. I have done quite now new things lately, will put soon. 

"MainActivity" handles the main screen and the video related stuff, except for the low-level stuff. I have noticed that when I use Inspire 1 or Mavic Air ( and probably Phantom 3) drones, we need a quite power full Android device to avoid video errors. So we need to look at performance and threading. Another ting is that it seems that there are threes that gets hanging after an exit, as some times I can run video just fine right after rebooting the Android device, but after restarting the app the video gets crap...

"MavLinkReceiver" handles all mavlink related stuff, so when a new Mavlink command is received it it parsed here. 
Then the actions is bilt in the DroneModel.  Unfortunately there are some waypoint stuff in this file also, that is quite confusing, and not working at the moment. If someone could fix the waypoints that would be great. Remember DJI got some strange demand for minimum and maximum distance and minimum number og waypoints etc. 

The rest of the files are helper files, doing small dedicated stuff as can be seen. 
I am about to add a generic AI screen that every one can modify to their own likings. 
I use this mostly for AI research using Python on a Linux host computer to control the drone from live video.  

The layout is at the moment tuned for DJO SmartControll screen layout. Some more dynamic layout would be fine. 
On the CrystalSky it looks a bit odd.

The1only, 2020-11-11:
I have added some experimental DJI TimeLine usage lately, to test it out. At the moment I use this to do takeoff to x meters, as DJI do not support that. In the current code I have created that function using very low-level virtual stick movements to climb while holding position. It work well, but the TimeLine might make is simpler and hourly better in wind. I will add all this code later to day.

The1only, 2020-11-11:
If anyone know how to make a plugin interface to the software please let me know. We need a plugin for the AI, such that every one can make their own AI stuff without getting involved in RosettaDrone2 it self...

lukelu0520, 2020-11-13:
@The1only , thanks for your guide. [The module of validateTranscodingMethod in MainActivity.java](https://github.com/The1only/rosettadrone/blob/fc428874917c44f70c44534dbd983c6b5a8fe9bd/app/src/main/java/sq/rogue/rosettadrone/MainActivity.java#L751) doesn't include the case of MAVIC_MINI. Is it intended because mini doesn't need old handling ?

Hope to hear your expertise.

Best,
-- Luke

kripper, 2023-02-15:
Hi @The1only and team,

I have good news.

1.-  I just finished testing, reviewing and merging all forks, added support for native MAVLink waypoints and many other features, fixed hundreds of bugs, implemented feature requests, cleaned a lot of code, etc.

It's time to upload all my changes to the master branch and respond and close the issues.

Since this is project has been dead for over a year and I know you have no time, please give me full access to this repo since I will be the active maintener for some time in the future.
If not possible, I will create a "Rosetta 3" forked project, but I prefer to keep the development centralized.

In the future, please let's start commiting atomic changes immediately and apply [good GIT practices](https://about.gitlab.com/topics/version-control/version-control-best-practices/).

For example, the merge of the CommandManager was a good move, but I noticed you had to revert it because we are not up to date.
The codebase still needs a lot of refractoring and cleaning, but we first have to get up to date.

2 .- We are connecting Mavlink with DJI, the best flying drones considering features/price.
After cleaning the mess, I will contact the other opensource communities (ArduPilot, DroneCode, etc) and ask them to include "Rosetta 3" to their list of "member" projects.

danammeansbear, 2023-02-15:
Please reach out to me as I want to help maintain this project. I would
consider myself midlevel software engineer and would like to help out where
possible. Id like to be a bay scrum master for this project if given some
guidance!!
*Adam Dabdoub*
Software Developer
c: 513.886.0301

kripper, 2023-02-16:
@The1only I still have no access to close issues or merge pull requests. My user is "kripper"

danammeansbear, 2023-02-16:
@kripper once you have been added, please start assigning issues/bugs/tests
to me and I will start working on it.
*Adam Dabdoub*
Software Developer
c: 513.886.0301

kripper, 2023-02-16:
@danammeansbear sure

kripper, 2023-02-16:
@The1only you got confused and added Adam instead of me :-)

I can fix it once you give me access.

The1only, 2023-02-17:
Oppps hmm. Send me you’re github name and I’ll fix it.Best regardTerje Nilsen9Tek ASNorway

The1only, 2023-02-17:
Fixed !!! I belive :-)

kripper, 2023-02-17:
@The1only yes, thanks. Master branch is up to date now.
Please everybody switch back to master and report issues.

chron0, 2023-02-17:
++ thanks for keeping this project alive and moving it forward. being able to use qtgroundcontrol over mavlink with dji hw is <3

kripper, 2023-02-17:
SUBJECT: Precision landing

Hi team,

WayPoint missions are working fine now with QGC using the new MissionManager class (using VirtualSticks).
Next, I will implement precision landing using Python and OpenCV.
There are a lot of python scripts around on Github but some of you probably have also been playing with precision landing in the past.
Does anybody want to share their existing precision landing implementation to Rosetta?

I'm available for reviewing, cleaning and integrating the code into master so we can all maintain and improve it after.
My recommendation is to always share your code and stay up to date on the same codebase.
Otherwise personal efforts and implementations get obsolate and start to conflict with upstream implementations which sooner or later will be superior and well maintained by others.

The1only, 2023-02-18:
I got object seek/detection and following/tracking using openCV and a trained «resdet» AI and Rosetta with som kalmanfilters etc.I am considering releasing it. However the trained AI I can not release, but it’s not that hard to make something for landing.Best regardTerje Nilsen9Tek ASNorway

kripper, 2023-02-18:
I'm thinking in integrating the precision landing Algorithm using Aruco from this repo:
https://github.com/tizianofiorenzani/how_do_drones_work

kripper, 2023-02-18:
BTW, have you tried MAVSDK?
I'm running it locally on the smartphone listening on a UDP port.
Rosetta connects to the port (MAVSDK acknowledges the connection), but somehow the mavlink commands sent by MAVSDK are not received by Rosetta.

The1only, 2023-02-18:
Yes used it with python all the time. There was a trick, I’ll check my code.Best regardTerje Nilsen9Tek ASNorway

The1only, 2023-02-18:
No sorry I used mavutil:#!/usr/bin/env python3# -*- coding: utf-8 -*-"""Created on Thu Jul 30 16:12:11 ***@***.***: smartdrone"""# #########DEPENDENCIES#############from dronekit import Command, connect, VehicleMode,APIException,LocationGlobalRelative,LocationGlobalimport timefrom   pymavlink import mavutilimport threadingimport mathimport numpy as npfrom AI_module import Modesfrom contextlib import contextmanagerfrom nose.tools import assert_equal# ########GLOBALES#################        # ########FUNCTIONS#################class drone_controller:    def __init__(self):        self.g_elev = 1500        self.g_yaw = 1500        self.g_pich = 1500        self.g_roll = 1500        self.g_sw1 = 1000        self.g_sw2 = 1000        self.g_sw3 = 1000        self.g_AImode = Modes.NONE        self.modeoverride = False        self.takeoffheight = 10        self.t_stop = False        self.airborn = False        self.takeoffheading = -1        self.just_seen=[]               # Place holder...        self.sitl = None        self.command_OK = False        print('Controler Initialized...')            # We go ahead and call our observer once at startup to get an initial value...    def addObserverAndInit(self, name, cb):        self.vehicle.add_attribute_listener(name, cb)    #----------------------------------------------------    def message(self, name, message):        self.flightstatus = message        self.flightstatusname = name        self.flightstatusready = True        # Create a message listener for all messages.    def receivecontrol(self, name, message):        self.g_elev = message.chan1_raw        self.g_yaw = message.chan2_raw        self.g_pich = message.chan3_raw        self.g_roll = message.chan4_raw        self.g_sw1 = message.chan5_raw        self.g_sw2 = message.chan6_raw        self.g_sw3 = message.chan7_raw        if self.modeoverride == False:            if message.chan8_raw == 1000:                self.g_AImode = Modes.NONE            elif message.chan8_raw == 1100:                self.g_AImode = Modes.LINETRACKING            elif message.chan8_raw == 1200:                self.g_AImode = Modes.POLETRACKING            elif message.chan8_raw == 1300:                self.g_AImode = Modes.LINELOOKING    # Complete status...    def completefunc(self, name, message):                       print("1) Reseived: %s -> %s" % (name,message))        if message.command == mavutil.mavlink.MAV_CMD_NAV_TAKEOFF and message.result == 0:            self.command_OK = True                    elif message.command == 84 and message.result == 0: # Set position local NED...            self.command_OK = True#            print("1) Reseived: %s -> %s" % (name,message))                    elif message.command == 86 and message.result == 0: # Position Conditional YAW...            self.command_OK = True#            print("2) Reseived: %s -> %s" % (name,message))                    elif message.command == mavutil.mavlink.MAV_CMD_DO_SET_SERVO and message.result == 0:            self.command_OK = True#            print("3) Reseived: %s -> %s" % (name,message))    #----------------------------------------------------    # @contextmanager    # def assert_command_ack(self, command_type, ack_result=mavutil.mavlink.MAV_RESULT_ACCEPTED, timeout=10):    #     """Context manager to assert that:    #     1) exactly one COMMAND_ACK is received from a Vehicle;    #     2) for a specific command type;    #     3) with the given result;    #     4) within a timeout (in seconds).    #     For example:    #     .. code-block:: python    #         with assert_command_ack(vehicle, mavutil.mavlink.MAV_CMD_PREFLIGHT_CALIBRATION, timeout=30):    #             vehicle.calibrate_gyro()    #     """        #     acks = []    #     def on_ack(self, name, message):    #         if message.command == command_type:    #             acks.append(message)        #     self.vehicle.add_message_listener('COMMAND_ACK', on_ack)    #     yield        #     start_time = time.tim       # Define vehicle listner, and forward to controller...    #     while not acks and time.time() - start_time < timeout:    #         time.sleep(0.1)183    #     self.vehicle.remove_message_listener('COMMAND_ACK', on_ack)        #     assert_equal(1, len(acks))  # one and only one ACK    #     assert_equal(command_type, acks[0].command)  # for the correct command    #     assert_equal(ack_result, acks[0].result)  # the result must be successful    def connect(self, connection_string):        print("Connecting...")        if not connection_string:            from dronekit_sitl import SITL, start_default            self.sitl = SITL() #           self.sitl.download('solo', '2.0.20', verbose=True) #           sitl_args = ['-I0', '--model', 'quad', 'param load ./solo.param', '--home=60.4094,10.4911,0,45']            self.sitl.download('copter', '3.3', verbose=True)            sitl_args = ['-I0', '--model', 'quad', '--home=60.4094,10.4911,0,0']            self.sitl.launch(sitl_args, await_ready=True, restart=True)                  #    connection_string = 'tcp:127.0.0.1:5760'        #    sitl = start_default()            connection_string = self.sitl.connection_string()            print("Simulate mode!!!!")        self.vehicle = connect(connection_string, wait_ready=True)        self.vehicle.wait_ready('autopilot_version')                # We need a position for the Pole tracker...        self.just_seen = self.vehicle.location.global_relative_frame     #   self.vehicle.add_attribute [... cut ...]     else:            if newang > (math.pi*2): newang = newang - (math.pi*2)            if newang < 0: newang = newang + (math.pi*2)                    return newang        def get_bearing(self,lat1,lon1,lat2,lon2):        dLon = lon2 - lon1        y = math.sin(dLon) * math.cos(lat2)        x = math.cos(lat1)*math.sin(lat2) - math.sin(lat1)*math.cos(lat2)*math.cos(dLon)        brng = np.rad2deg(math.atan2(y, x))        if brng < 0: brng+= 360        return brngBest regardTerje Nilsen9Tek ASNorway

danammeansbear, 2023-02-20:
got a new development machine, give me time to set everything up on Mac. will let you know.

kripper, 2023-02-21:
@The1only can we rename the project to "Rosetta Drone" (without the version number) or just "Rosetta"?

kripper, 2023-02-21:
I just found out that Rosetta was not supporting MAVLink 2.
I'm going to implement it now.

The1only, 2023-02-21:
We can, if you feel like it. Be aware the origional rosettadrone 1 (named rosettadrone) that gave up and handed it to me years ago. We do not whant to confuse people as that one is totally different.Ps: you do a greate job by the way, I support you 100%Best regardTerje Nilsen9Tek ASNorway

chron0, 2023-02-21:
I would advise against using just "rosetta" - that will conflict with the macos x86 emulation system thats also called rosetta and create unnecessary noise when searching for issues in the future. so, I'd suggest keeping rosettadrone or coming up with a new unique name.

kripper, 2023-02-21:
I'm contacting the other Rosetta Drone repo owners on Github to link to this repo.
It would be good to transfer the owner of this repo to a official "RosettaDrone" user.

I've added MAVLink 2 support on master branch and tested it with QGC.
The version number is now hardcoded = 2.

The1only, 2023-02-21:
The difference from the other Rosetta Drone repo is so huge, we would still need to call this Rosetta Drone V2.x, or better V3.x to avoid confusion as there now again is so many changes.I do not quite see the problem.Best regardTerje Nilsen9Tek ASNorway

kripper, 2023-02-24:
The other repos on GitHub are now official unmaintained and linking to us.

I need help cleaning the video code.

I wonder why we are not just sending the raw data received from the SDK to a destination UDP port to process it outside Rosetta using ffmpeg (instead of decoding it inside Rosetta with the native helper).
Would it be possible?
I tried it shortly with a mini SE, but wasn't able to decode the video on the end point with ffmpeg, even when ffprobe recognized it as h264 and it's size.
Maybe ffmpeg needs special options?

When trying with the original RTP stream "packetized" by Rosetta, gstreamer works but the video looks as if it had encoding/decoding errors. I couldn't decode or detect the stream with ffmpeg or ffprobe. I could also be a performance problem, so I will try to just decode single frames. I want it for computer vision.

@The1only could you maybe confirm if your latest video code is in master branch?

kripper, 2023-02-24:
I see. You got too long UDP packets when sending them out without decoding first. Here is your old post: https://github.com/The1only/rosettadrone/issues/8

I'm just worried about the decoding and encoding (on Rosetta) and then decoding again on the AI end point.
Maybe Rosetta could send already decoded frames as raw to the AI script in a given resolution and frequency.
Specially when the AI script is running locally on Android.

kripper, 2023-02-24:
I created this issue to discuss the idea of doing the decoding outside Rosetta:
https://github.com/The1only/rosettadrone/issues/122

The1only, 2023-02-24:
Let me check, there was no issues vith video on Mavic Pro 2…. However there was some performance issues with gstreamer. This is whay we used Nvidia GPUs when we compiled gstreamer. I belive, sometime there is a loot of I frames that makes the decoder strugle…Then again this was years ago now. Everything has changed.Best regardTerje Nilsen9Tek ASNorway

The1only, 2023-02-24:
If you look at the video I made, the overall latency was supprisingly short (alt least to me). So no effort was made to improve it future. However flying fully autonomously with AI and video got issues when we passed approx. 5m/s when fine navigation/tracking. This is why I had to add a kalmanfiler in the python AI code.Best regardTerje Nilsen9Tek ASNorway

kripper, 2023-03-05:
> If anyone know how to make a plugin interface to the software please let me know. We need a plugin for the AI, such that every one can make their own AI stuff without getting involved in RosettaDrone2 it self...

Done.
https://github.com/The1only/rosettadrone/commit/93cf9efd57a7f31b14e01bcb7f92cb9441b80cef

You can now move your custom code to a class that extends the Plugin class.
This way we can keep the MainActivity clean.

danammeansbear, 2023-03-06:
> 93cf9ef

I have an event based architecture I am working on with a desktop and
mobile app. Using rabbitMQ, Blazor and unity.
Once complete I will be uploading it into my repo for testing as a demo
project.
*Adam Dabdoub*
Software Developer
c: 513.886.0301

kripper, 2023-03-19:
> with som kalmanfilters

I believe I need to implement EKF.

The1only, 2023-03-19:
EKF to do what?

kripper, 2023-03-19:
> EKF to do what? I normaly try to stay with KF.

I'm using visual pose estimation of Apriltags for precision landing (I'm preparing to code to release).
I see at least 3 problems to address:

1) how to get the attitude of the drone. a) Only use visual pose estimation, b) only use the drone positioning system or c) combine both.

If the error of the positioning system is too big, maybe it makes sense to only use vision.

2) the visual estimated pose will not be the current pose because of latency. This means, we would have to do corrections on the model using information from the past, to be able to use the model in the future (prediction). I believe EKF provides this solution.

3) which frame model to use? If the positioning system is accurate, a absolute frame model would be the best. Otherwise, if we use vision, maybe a frame relative to the Apriltag is convenient.

kripper, 2023-03-22:
Hi @The1only, can you please take a look at https://github.com/The1only/rosettadrone/issues/132?

I'm interested in your experience with KF.
At least on my model, getVelocityX and GPS information are useless.

kripper, 2023-04-08:
Closing. Development discussion continues here:
https://github.com/RosettaDrone/rosettadrone/issues/132

closed 2023-04-08 by kripper