RosettaDrone/rosettadrone #22: No control

ryansabin, 2020-08-31:
I have RosettaDrone running on the DJI SmartController with a Mavic 2 Pro. I have QGroundControl running on Ubuntu Linux.

With the exception of a few crashes it seems to be running OK. The video seems to lag and become sticky. I even have the video streaming to the QGroundControl station.

I am having issues trying to control the drone from QGroundControl. I have tried Go To Waypoint, Uploading a simple flight mission. I cannot control it with the sticks in QGroundControl either.

Control from the RosettaDrone app is working fairly well. With the exception of crashes every once in a while.

I have the time and experience to work to resolve these issues. Please let me know what I can do to get this up and running.

The1only, 2020-09-01:
Hmm it some time chrasjes at start, it’s an open-gl issue it seems, but I also use the Smart controller an it never chrases while running.

The joystick should work, I have to check that.
Use QGC 3.5 the 4.x is bad, they work on it. The QGC team has seen the same thing now.

I use DroneKit for control, missions still got issues. Goto is not tested, DJI does not support it so I hade to write it from schrach, using low-level controlls.

I try to pick one snd one isse, at the momen video on non OcySync systems is prioritized.

Feel free to add push requests and fixes. 

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2020-11-14:
There are still issues with some of the goto commands and waypoint... will be fixed soon...

marcfeather, 2021-06-21:
The1only,

Thank you for this project allowing us to manage the DJI product with QGroundControl. In my testing the Mavic 2 Pro Smart Controller video feed I am not able to get a video feed from the Smart Controller to QGroundControl. 

I have successfully tested this on multiple phones and tables with success but impossible to find a reason for the SC not sending the feed to QGC 4.1.3.

Any ideas on where to start?

Regards

The1only, 2021-06-21:
Check off the «super low latency» flag in QGC, and make sure to use a computer and network that can handle the trafik.

I controll the drone that way all the time, super low latency (0.5s total fram event to decoded in Python using OpenCV, even lower in QGC) , and no chrashes.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2021-06-21:
Yes the problem is normally wrong version of the compile tools. You will then loose video, do not know why.

Best regard
Terje Nilsen
9Tek AS
Norway

closed 2023-02-24 by kripper