RosettaDrone/rosettadrone #23: Video out from RosettaDrone 2 does not work on older non OcuSync unites

The1only, 2020-09-03:
The RosettaDrone 1 ONLY works with the older video systems, used on Phantom etc. My plan was to skip all older systems. Now I see that new systems like Mavic Air and Mavic Mini still uses the the old video format. And I am getting a loot of request to add video support for these to RosettaDrone 2. This is a complicated task that involves JNI C based FFMPEG and GStreamer functions. However the customers are always right so I am on my way deep in to the inner makings of the DJI video format. I do expect higher latency on these systems as I need to decode the video stream using FFMPEG and then re-encode the video using gStreame. However I will get there...!!!  If someone actually fully understands these JNI-C classes, please feel free to jump inn.

The1only, 2020-09-03:
Can any one confirm or add to this so we get it updated, these files are years old:

public int getIframeRawId(Model pModel, int width) {
        int iframeId = R.raw.iframe_1280x720_ins;

        switch(pModel) {
            case PHANTOM_3_ADVANCED:
            case PHANTOM_3_STANDARD:
                if (width == 960) {
                    //for photo mode, 960x720, GDR
                    iframeId = R.raw.iframe_960x720_3s;
                } else if (width == 640){
                    iframeId = R.raw.iframe_640x368_osmo_gop;
                } else {
                    //for record mode, 1280x720, GDR
                    iframeId = R.raw.iframe_1280x720_3s;
                }
                break;
            case INSPIRE_1: {
                DataCameraGetPushStateInfo.CameraType cameraType = DataCameraGetPushStateInfo.getInstance().getCameraType();
                if (cameraType == DataCameraGetPushStateInfo.CameraType.DJICameraTypeCV600) { //ZENMUSE_Z3
                    if (width == 960) {
                        //for photo mode, 960x720, GDR
                        iframeId = R.raw.iframe_960x720_3s;
                    } else if (width == 640){
                        iframeId = R.raw.iframe_640x368_osmo_gop;
                    } else {
                        //for record mode, 1280x720, GDR
                        iframeId = R.raw.iframe_1280x720_3s;
                    }
                }
                break;
            }
            case Phantom_3_4K:
                switch(width) {
                    case 640:
                        //for P3-4K with resolution 640*480
                        iframeId = R.raw.iframe_640x480;
                        break;
                    case 848:
                        //for P3-4K with resolution 848*480
                        iframeId = R.raw.iframe_848x480;
                        break;
                    case 896:
                        iframeId = R.raw.iframe_896x480;
                        break;
                    case 960:
                        //DJILog.i(TAG, "Selected Iframe=iframe_960x720_3s");
                        //for photo mode, 960x720, GDR
                        iframeId = R.raw.iframe_960x720_3s;
                        break;
                    default:
                        iframeId = R.raw.iframe_1280x720_3s;
                        break;
                }
                break;
            case OSMO:
                if (DataCameraGetPushStateInfo.getInstance().getVerstion() >= 4) {
                    iframeId = -1;
                } else {
                    iframeId = R.raw.iframe_1280x720_ins;
                }
                break;
            case OSMO_PLUS:
                if (width == 960) {
                    iframeId = R.raw.iframe_960x720_osmo_gop;
                } else if (width==1280) {
                    //for record mode, 1280x720, GDR
                    //DJILog.i(TAG, "Selected Iframe=iframe_1280x720_3s");
                    //                    iframeId = R.raw.iframe_1280x720_3s;
[... 173 lines cut ...]
                        DJILog.i(TAG, "Selected Iframe=iframe_1920x800_wm620");
                        iframeId = R.raw.iframe_1920x800_wm620;
                    } else if (width == 1920 && height == 1024) {
                        DJILog.i(TAG, "Selected Iframe=iframe_1920x1024_wm620");
                        iframeId = R.raw.iframe_1920x1024_wm620;
                    } else if (width == 1920 && height == 1088) {
                        DJILog.i(TAG, "Selected Iframe=iframe_1920x1080_wm620");
                        iframeId = R.raw.iframe_1920x1088_wm620;
                    } else if (width == 1920 && height == 1440) {
                        DJILog.i(TAG, "Selected Iframe=iframe_1920x1440_wm620");
                        iframeId = R.raw.iframe_1920x1440_wm620;
                    }
                }
                break;
            case PHANTOM_4_PRO_V2:
            case PHANTOM_4_RTK: {
                iframeId = -1;
            }
                break;
            default: //for P3P, Inspire1, etc/
                iframeId = R.raw.iframe_1280x720_ins;
                break;
        }
        return iframeId;
    }

The1only, 2020-09-11:
Ahh sorry that will not work, I have the IP address sett static for testing. Still some issues with Binding to service...

gsenkowski, 2020-10-16:
> The new branch Video_Work got half way support or video out on older systems.
> 
> Video can be decoded with: sudo gst-launch-1.0 -vvv udpsrc port=5601 ! queue ! h264parse ! decodebin ! videoconvert ! autovideosink
> 
> Or if you run this in a shell on the local PC the QGC will work:
> 
> sudo gst-launch-1.0 -vvv udpsrc port=5600 ! queue ! h264parse ! decodebin ! videoconvert ! videorate ! x264enc ! rtph264pay ! queue ! udpsink host=127.0.0.1 port=5601
> 
> Set QGC to UDP at port 5601.
> 
> The remaining problem is the GStreamer in Rosetta is old and not compatible, so it chrashes. I am working on it.
> The line abou takes the raw h.264 and ether decode it or transcode it to rtp h.264...
> 
> Pleas try this on all systems so we can verify...

Tested this with Mavic Mini, works for me.

The1only, 2020-10-16:
I belive it shuld fully work now, if you can test with QGC.

Best regard
Terje Nilsen
9Tek AS
Norway

closed 2023-02-24 by kripper