RosettaDrone/rosettadrone #25: Websocket or other protocols for web browser communication

lukelu0520, 2020-11-08:
Dear rosetta drone friends,

The app is really helpful and fantastic, allowing many android users fly a DJI drone with a Mavlink-based ground control station. So far it works well with QGroundControl. QGroundControl requires local installation, so we were wondering if rosetta drone could work with a browser.

For example, are we able to use js and websocket, send / receive mavlink messages to / from rosetta drone.

Hope to hear your expertise.

Best,
-- Luke

The1only, 2020-11-09:
I normally set up a virtual server in the cloud to enable this. It is posible to comunicate with this virtual pc, then forward the socket to whatever.

lukelu0520, 2020-11-09:
@The1only thanks for your kind reply. We'll start by setting a MAVRos on a cloud server to see if we are able to receive mavlink data from a mobile device (is connected to DJI remote controller) installed rosettadrone. 

Best,
-- Luke

closed 2020-11-10 by lukelu0520