RosettaDrone/rosettadrone #27: Waiting for video in QGroundControl

lukelu0520, 2020-11-11:
Dear rosettadrone friends,

In QGroundControl, we are able to take off, landing, got summary of drone parameters by external GCS.
However, the video panel is always showing "waiting for video", even though we've set Video Source to UDP Video Stream (tried both H.264 and H.265), and set UDP Port to 5600.  Port forward is also setup to make sure port 5600 is properly routed.

We were wondering if someone got the same issue with us. Our drone is DJI Mavic 2 Zoom. QGroundControl is running in Ubuntu 18.04 LTS.

Hope to hear your expertise.

Best,
-- Luke

The1only, 2020-11-14:
That should work perfect, check the firewall settings and try using netcat to trace the udp stream. I uses that setup every day, with very low latency.

The1only, 2020-11-14:
Ps: there are some issues with some versions of gradle, do NOT update anything untill you can make working code, then update gradually.

lukelu0520, 2020-11-14:
@The1only , thanks for your support. Which gradle version are you using for the best performance ? We can downgrade to that working version.

Thanks,
-- Luke

The1only, 2020-11-14:
I use 3.5.3 and 6.3, the video with the Mavic 2 zoom is supose to be perfect.

lukelu0520, 2020-11-14:
Update: We disabled "Enable Video feed", and video becomes perfect. It's interesting, Do you know the reason ?

Best,
-- Luke

The1only, 2020-11-14:
Hmm hav not tested that in a while, disable the secondary or primary video feed?

lukelu0520, 2020-11-14:
Thanks for your prompt reply! We disabled all of them in Video of Setting.  (we expected video would disappear, but turns out it works perfectly.)

The1only, 2020-11-14:
I added variables to the config file, maybe the system was still using an old config file with the new settings, and when flipping a new file was written and the problem fixed....  must remember that for later updates.

The1only, 2020-11-14:
Just passed a majore computer vission AI project, flying live with a DJI Mavic and a CNN AI running on an eGPU on the QGC computer. The drone used Rosettadrone video and telemetry for control doing complex movements... happy days...

lukelu0520, 2020-11-14:
@The1only , thanks for your support. After making mavic 2 zoom video working, we are now working on mavic mini.
We added mavic mini as a new case in validateTranscodingMethod and received some juices, however, the video is noisy signal. (seems like a bad encoding/decoding) 
Could you tell us which config file (or the portion of code) we could check in order to make mavic mini working ?

The1only, 2020-11-15:
First try rebooting EVERYTHING, and try fresh. If that does not help, then it gets complicated, we will have to work our the NAILs and framing... it can take a few weeks to work out.  I do not have a mini my self.

The1only, 2020-11-15:
We might at some point have to reintroduce gStreamer to be able to set transfer quality again

lukelu0520, 2020-11-15:
@The1only , thanks for the advice. Bad news is rebooting doesn't help, MINI's video stream shows bad encoding on the screen of our Android device.

To solve the issue, we were wondering if we could bypass H264 encoding inside rosetta drone; directly send raw DJI video format (assuming it is in YUV format) out for the screen and stream it out via UDP. 
Then we could use gst-launch-1.0 to receive the video and do the H264 encoding on more powerful desktop.

Do you think this approach is doable and is reasonable to execute ?

Best,
-- Luke

The1only, 2020-11-15:
Aha if there is bad video on the android device that is good news, as the SDK IS suppose to support that. Only the exporting is our responsebility.

Now take the video example made by DJI for git hub, see if that works, it should. Then from that start to compare.

The1only, 2020-11-15:
I bought a Mavic Mini today, and I see the same problem. On the Android device I have perfect video, but on QGC I got crap... I also got an Mavic AIR I will retest this drone, I hade belived they are similar....

lukelu0520, 2020-11-15:
It's glad to hear you got a Mini and it works on your android device. (my android device got crap.)

By reviewing the ffmpeg part (i.e. when mode=0) of dji_video_jni.c, we don't find there is a significant difference from [DJI's Android Video Stream Decoding Sample's  dji_video_jni.c ](https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/master/android-videostreamdecodingsample/jni/dji_video_jni.c) 

We can not see few .c / .h files in rosetta's folder, e.g. cmdutils.c, cmdutils_common_opts.h, ffmpeg_filter.c, ffmpeg_mod.c, ffmpeg_opt.c, and Application64.mk.

We are gonna to try DJI's Android Video Stream Decoding Sample to see if it could output YUV data from Mini.

Best,
-- Luke

The1only, 2020-11-16:
Hmm, on my system I got perfect video on the Android device. Remember that the ffmpeg stuff is used for the encoding, while the android device screen is using the decoding from the sdk. If the device video is crap, then there is no need to look at the C code.

lukelu0520, 2020-11-17:
I remember your Android device is Samsung Note20 phone. What's the android version on the phone ?
We'll see if we could get the same phone and system version with you.

lukelu0520, 2020-11-17:
@The1only , thanks for your information. Update: We are on the same page now. Got perfect video on the Android device.
Again, turn off and turn on "Enable Video Feed" solve the problem.

However, still got crap video via UDP.

The1only, 2020-11-17:
The on and off problem seems to be a sync issue, I hade the same and rebooted the drone and it got fine. But you're method is better. After we get the video to work on QGC then I will look into this.

The1only, 2020-11-17:
What QGC version do you use that works???

lukelu0520, 2020-11-17:
We are using QGC 4.0.11 and gst-launch-1.0 version 1.14.5 (GStreamer 1.14.5).

Thanks for pointing sync issue out. So far we change the default setting of mExternalVideoOut and pref_enable_video to false.
But we were wondering if there is a better way to fix the sync issue. May be auto-set mExternalVideoOut to false and then re-set to true when TextureView is set or wait 1 sec after onResume is called ?

Hope to hear your expertise.

lukelu0520, 2020-11-19:
Update: problem solved, we fixed gstreamer pipeline issue; we could live stream MINI's video on both QGC and gstreamer now.

There is only issue remain -- the sync issue, which requires us disable "Enable Video Feed" every time when finishing the use of the app.

We'll dig into the problem to see we could make a contribution.

Best,
-- Luke

The1only, 2020-11-19:
Do tell, what and how did you fix the gStreamer pipeline?

lukelu0520, 2020-11-19:
@The1only , sure, regards to the gstreamer issue, please check the [PR#9180](https://github.com/mavlink/qgroundcontrol/pull/9180) we submitted to mavlink/qgroundcontrol.

Regards to the video sync issue on rosetta drone, let us know if any suggestion or guide to resolve it.
Hope to find a better way to fix it !

Best,
-- Luke

The1only, 2020-11-19:
So PR#9180 actually made a difference in this case?

As an experiment we can try always start with video off, and then set the selected vodep state after alll system is up.

lukelu0520, 2020-11-20:
No, this is why I didn't post the PR here until you asked for that.

We recently find MINI's h264parse and rtpsocket might have k-frame or header error. This minor issue propagates and impact later stages of application. But it is out of the topic. Maybe I should close the thread.

Thanks for your active support!

--Luke

kripper, 2023-02-25:
Hi @lukelu520,

I want to fix the problem with the MINI's RTP video stream.
I'm experiencing the same problem as you with my MINI SE.
Does it really fix the problem to disable and re-enable the video feed setting? I'm doing it, but it doesn't fix for me.
What command line are you using for gstreamer?

I found out that 'isKeyFrame' is always `false` in `onDataRecv()`. Maybe thatt is causing the sync problem.

kripper, 2023-04-08:
See: https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/edit#heading=h.q9mp6yjsfxe5

closed 2023-04-08 by kripper