RosettaDrone/rosettadrone #28: launch in android studio emulator

lukelu0520, 2020-11-13:
Dear rosettadrone friends,

We'd like to use emulator to develop the app efficiently. We've tried multiple virtual devices s.t. pixel 2 / 3a API 30_x86. We are able to install the app in these virtual devices, but can not launch the app.

The error message is:
`A/libc: Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xedcd3000 in tid 4034 (ue.rosettadrone), pid 4034 (ue.rosettadrone)`

We were wondering if any one can launch rosettadrone in an emulator successfully.  
Hope to hear your expertise.

Best,
-- Luke

The1only, 2020-11-14:
The new M1 from Apple is a arm64, and thus will be able to run Android native. So with Android Studio on the new Apple product we might get a perfect Android development plattform. Maybe we can even run Rosetta drone2 as a full screen PC app, we do not even need Rosetra2 from apple 😄

lukelu0520, 2020-11-14:
@The1only , thanks for your kind reply. Are you mainly using Mac to develop rosetta drone2 ?
We are using ubuntu 18.04. If you think Mac is a better option, we'll try to Mac for the development of rosetta drone 2.

Best,
-- Luke

The1only, 2020-11-14:
I used a Mac before, now I also uses Ubuntu, I have ordered one of the new Mac M1's that runs  ARM64. However they are not released yet in Europe. They have the potential to be the perfect development plattform, but we do not know.

Mac or Ubuntu makes little difference. I can not run the simulator with any og them, as they all uses Intel/AMD CPU's that is a totally different arcitecure. And the emulation got some issues. 

At the moment I uses Samsung Note20 phone, I used to use an Azus M... I also uses the DJI Smart Controller and the CystalSky all works great.

I need Cuda with Tensorflow for my AI app, so for me only Ubuntu can handle this, Apple no longer support Nvidia. I also need eGPU to run my CNN real time. The new Apple comes with a prommesing AI hardware module, that will be interesting to test.

BkMak16, 2022-01-24:
Did you manage to get the Mac, is it as promised? LOL. 

I might have to get my own if so, maybe run the AI on an external GPU

The1only, 2022-01-24:
With the Mac M1 it would run, sort off.
The emulator does not support the USB direct access so you kan not connect to the controller.
Too bad.

Best regard
Terje Nilsen
9Tek AS
Norway

kripper, 2023-02-21:
No AoA support?

Yesterday I released a DummyProduct class.
You can now test Rosetta without connecting to a drone.

Testing directly on the smartphone is faster in my case.

closed 2023-02-21 by kripper