RosettaDrone/rosettadrone #30: FTP Capabilities

lilfish, 2020-11-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

At the moment I'm working on a project that needs to analyze photos using an AI model. The original idea is to let a program calculate the different waypoints, fly it there, take a high quality image, send it back to the program and repeat until done. At the moment I only see the possibility to send a live feed, which for our project is too little quality. For that reason I like to be able to take a picture in high quality and send it over to my program making use of the Mavlink File Transfer Protocol (FTP).

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

After digging into the Mavlink protocol I've noticed there are FTP capabilities which are not yet used in Rosetta Drone. I would like to ask if there is a possibility for this to be added.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I have pulled the Rosetta Drone project and got it working on a android phone using a marvic air (a marvic 2 doesn't seem to work, the app crashes after pressing "Open" (looking into this later)), and am trying to get the FTP capabilities working.

**Additional context**
Add any other context or screenshots about the feature request here.

\-

**Additional**

I am not sure if making a Feature request is appropriate or how, if I can make it work somehow, I can give it back to this github somehow. If it's not, feel free to delete this feature request.

The1only, 2020-12-17:
This is a very interesting topic for many cases. There are some challenges, however. DJI must turn off video to be able to get to the camera images, so one will lose live video while getting the pictures. And none of this functionality is added. The FPT is the easy part, the DJI side is a bit more complicated. But please feel free to jump in and add this I too teeds this.

lilfish, 2021-01-14:
Dear The1only,

I’ve been working to enable FTP for DJI with some stugle and some succes. We, me and my classmate, wanted to enable the FTP protocol for our Flower Power (school) project.

Beginning with the good news: It is possible to enable FTP using the MAVLink protocol.

Now the bad new:

  *   I could not run the newest version of rosetta drone (using SDK 4.14). The app closes when pressing the “open” button.

  *   FTP seems to crash after getting a second file, the app closes without any errors or warning. I’m trying to find out why the app crashes and fix this issue.
  *   I’m not allowed to fly the drone in my country anymore

I also took the liberty to change the code a bit. I’ve made 3 new classes which I call “manager classes”. The classes are:

     *   FTPManager
     *   CommandsManager
     *   MissionsManager

I’ve made these to clean up the big switch case used in the Mavlink receiver class.

At the moment I’m also trying to make a arrangement so that I’m allowed to keep the drone for a longer time so I can keep developping on Rosetta drone.

I currently have two questions for you:

  1.  How can I merge my changes with the latest rosetta drone app version (after testing ofcourse)
  2.  Is it possible to make rosetta drone a public and available app on the google store?

I hope to hear soon of you.

Kind regards,
Vincent V. (aka Fish)

======================================================================================================================================== E-maildisclaimer: Aan dit bericht, waaronder ook de eventuele bijlagen worden bedoeld, kunnen geen rechten worden ontleend. Dit bericht is uitsluitend bestemd voor de geadresseerde en bevat persoonlijke en vertrouwelijke informatie. Het bekend maken, kopieren en verspreiden van een bericht dat niet voor u bestemd is, is niet toegestaan. Als u dit bericht per ongeluk heeft ontvangen, verzoeken wij u vriendelijk het direct te vernietigen en de afzender te informeren. Twijfelt u over de juistheid of volledigheid van dit bericht? Neem dan contact op met de afzender. ======================================================================================================================================== This message and any associated attachments cannot be deemed as legally binding under any circumstances. This message is intended for the exclusive use of the person(s) mentioned as recipient(s) and contains personal and confidential information. Directly or indirectly copying, disclosing, distributing, printing, publicising and/or in any way using this message or any part thereof by any means is strictly prohibited if you are not the intended recipient(s). If you have received this message in error, please notify the sender and delete this message from your system immediately. If you are in doubt regarding the correctness and/or completeness of this message, please contact the sender.

The1only, 2021-01-14:
Try using the "Work" branch. 

Try to make pull request when ever you made a change and tested it, do not way until you have a loot of changes, then it will be harder to merge and approve. 

Yes it can be put on App Store etc., it is a BSD3 and MIT so you are required to rename the project when you make an app. 
The name RosettaDrone is reserved, other than that you can use it. Also remember there are some libraries that require you to have an about button that lists all the licenses used, as IOS and Android also have. 
So make a name, change the icon/logo in the boot screen, and off you go...
But you then holds all liabilities for that. :-)

The1Only

lilfish, 2021-02-02:
@The1only , I made on pull request containing the CommandsManager changes, but I'm unsure if I did it correct. Could you check the pull-request and tell if it is correct and I can proceed like this or if it should be done differently.

The1only, 2021-02-02:
Seems you made the pull request against the master and not work branch ?? , as of this I will have to double merge the code, and that will take some time. In general I like you're idea of making a cleaner code. If you check out the work branch, patch that, and then push back with a pull request my work would be easier :-)

The1only, 2021-02-02:
The "work" will be merged with the "master" on the next release.

The1only, 2021-02-02:
The FTP thing... what are you trying to receive and how?  are you trying to download hires. images from the drone while in air? 
Have you verified that this is actually possible on DJI drones? I presume you then will loose video for a while when downloading?  I am very interesting in how you plan to get the files from the drone.

The1only, 2021-02-02:
Very cool,

Greate work, yes please use the «work» and I will accept it.

lilfish, 2021-02-02:
One issue I still have with the FTP is that it is possible to download one image, but the app crashes after downloading around 40%-50% of the second download.

The1only, 2021-02-02:
Ok that can be fixed, let’s get the first stepp into the code and we can all help. I like what you are doing. What drone do you use for testing?

lilfish, 2021-02-03:
> Seems you made the pull request against the master and not work branch ?? , as of this I will have to double merge the code, and that will take some time. In general I like you're idea of making a [...]

I just woke up and started everything up ready to make the new pull request, but I noticed that the commands Manager pull-request should point to "work":

> lilfish wants to merge 1 commit into The1only:work from lilfish:master

Could you double check this please? Otherwise I will just close the pull-request and remake it like I promised yesterday

lilfish, 2021-02-03:
> 
> Ok that can be fixed, let’s get the first stepp into the code and we can all help. I like what you are doing. What drone do you use for testing?

At the moment I have a mavic air 1 available, but about a month ago I tested it with the DJI Phantom 400 (which reminds me that it was/is not yet enabled in the code in the "MainActivity" Class at line 764. )

The1only, 2021-02-03:
I se you commitet to you're local bransh. I am very intereseted in you're work. Please also send a pull request to the official work bransh !!!!

lilfish, 2021-02-03:
> 
> 
> I se you commitet to you're local bransh. I am very intereseted in you're work. Please also send a pull request to the official work bransh !!!!

I'm having trouble with the branches. Currently I did something wrong when pulling the project I believe. I will need to re-write some code in a new class I made to make it work with the current rosetta drone work branch. Trying to get this fixed asap. Sorry for the hassle.

The1only, 2021-02-03:
Pull Request for command handler accepted, can you do the same for the FTP?

The1only, 2021-02-03:
I am not sure we actually gained anything with that last commit. Half the Command handler is still in the MAVlink file.

The1only, 2021-02-03:
If you'd like just send med the DroneModel.java, and I can merke it in. I had it, but you removed it :-)
I liked the FTP part

lilfish, 2021-02-03:
> 
> 
> If you'd like just send med the MainActivity.java, and I can merge it in. I had it, but you removed it :-)
> I liked the FTP part

Sorry about that, the original code is on a private gitlab repository owned by my school. I just came back from there with some good news, I'm allowed to keep the drone a little longer. 

Currently I'm adding the FTP manager. It is almost done but not tested yet.

lilfish, 2021-02-03:
Created a pull-request :)

The1only, 2021-02-03:
Accepted and added a few fixes, the code compiles, but not tested. «Work» is now untested.
But the new ftp is really exiting.

closed 2021-04-17 by lilfish