RosettaDrone/rosettadrone #32: no video streamer  show the screen

zhaocundang, 2020-12-03:
I　download master and video_work , complie ok , installed apk success, click start button , no video stream show

zhaocundang, 2020-12-03:
connect usb to joystick

The1only, 2020-12-03:
We got several good fixes for the video now. I believe we will get ALL DJI drone up working very soon.
I Will check-in in MAVIC PRO later today...

USB Joystick in QGC was tested quite some time back and was working?

gsenkowski, 2020-12-04:
I think the Joystick in QGC does not work with Rosettadrone 2 because of the handling of MANUAL_CONTROL messages https://mavlink.io/en/messages/common.html#MANUAL_CONTROL . 

https://github.com/The1only/rosettadrone/blob/137552a19f3267cc908ed9fdf82516ec66fc1cbd/app/src/main/java/sq/rogue/rosettadrone/MAVLinkReceiver.java#L282-L286

The bit mask on the end of line 284 `0b0000100000111000` actually masks out all the relevant values and leads to ILLEGAL PARAMETER errors in https://github.com/The1only/rosettadrone/blob/137552a19f3267cc908ed9fdf82516ec66fc1cbd/app/src/main/java/sq/rogue/rosettadrone/DroneModel.java#L1990 because no values for the FlightControlData object will be set.

The1only, 2020-12-04:
Not even on the Android device???

The1only, 2020-12-04:
Ahhhh shit... I was misled to believe that we could update the cradle version...
We must step back that one... Give me 5 minutes... I am so sorry.

The1only, 2020-12-04:
If you just change the Gradle file like this it will work again:

classpath 'com.android.tools.build:gradle:3.5.3'

The1only, 2020-12-04:
Hmm I belive you are right…. I have not tested that in year and it definitely seems wrong…

The1only, 2020-12-17:
This generally works now, still some issues with Gradel on some systems but we put that in as a separate issue.

closed 2020-12-17 by The1only