RosettaDrone/rosettadrone #36: Some examples using DroneKit and Python to fly the DJI drones...

The1only, 2020-12-03:
Two somewhat complex example that works is this:

ALWAYS test with Ardupilot SITL first, then with RosettaDrone2...

# Send a velocity command with +x being NOTRH of the drone.
def send_global_ned_velocity(self, vx, vy, vz, yaw):
    msg = self.vehicle.message_factory.set_position_target_global_int_encode(
        0,  # time_boot_ms (not used)
        0, 0,  # target system, target component
        mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT_INT,  # frame
        0b0000011111000111,  # type_mask (only speeds enabled)
        0, 0, 0,  # x, y, z positions (not used)
        vx, vy, vz,  # N, E, D velocity in m/s
        0, 0, 0,  # x, y, z acceleration (not supported yet, ignored in GCS_Mavlink)
        0,float(yaw))  # yaw, yaw_rate (not supported yet, ignored in GCS_Mavlink)
    self.vehicle.send_mavlink(msg)

# Send a velocity command with +x being the heading of the drone.
def send_global_ned_position(self, x, y, z, yaw, speed):
    msg = self.vehicle.message_factory.set_position_target_global_int_encode(
        0,  # time_boot_ms (not used)
        0, 0,  # target system, target component
        mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT_INT,  # frame
        0b0000111111111000,  # type_mask (only position enabled), ignore yaw for now...
        int(x*10000000), int(y*10000000), float(z),  # x, y, z positions (not used)
        speed, speed, 1.0,  # x, y, z velocity in m/s
        0, 0, 0,  # x, y, z acceleration (not supported yet, ignored in GCS_Mavlink)
        float(yaw), 0)  # yaw, yaw_rate (not supported yet, ignored in GCS_Mavlink)
    self.vehicle.send_mavlink(msg)

The1only, 2020-12-04:
PS: If you use the GOTO function in QGC you MIST move more than 50cm or it will ignore the command (a DJI thing).

danammeansbear, 2020-12-07:
This might sound trivial but could you make a basic tutorial for this ? once you have rosetta drone working and are able to do a QGC mission, where would i start to use this? my goal is to add a python VoiceBot to automatically handle take off, landing, and assignment of missions.

The1only, 2020-12-07:
I will, I am working on mission plans from QGC now, the goal is to have this working before Christmas.

lukelu0520, 2020-12-17:
Thanks @The1only ,it's a very useful tutorial on controlling drone using python.
Could you add an even simpler case: change arming status and takeoff the drone ?
We are trying to take off the drone using command line (with mavlink & mavros) but keep getting error, so are considering use python to solve the problem.

Best,
-- Luke

The1only, 2020-12-17:
All the information about using the Python DroneKit to control RosettaDrone2 is here:

https://dronekit-python.readthedocs.io/en/latest/

It will eventually be 100% Mavlink compatible, so wether you fly DJI or a Mavlink drone it should be the same.
There are some limitations at the moment, but work in progress.

The1only, 2020-12-17:
Ahhhh I know you're problem, you MUST unlock the system ( for safety) hitt the small lock button up on the right side, when it is white, THEN you can take off....  Just to prevent accidental takeoffs.

adpmdrones, 2020-12-17:
Hi All,

following a very simple python script that connects to RosettaDrone and print out some vehicle parameters:

Usage:
 **python rosetta_drone.py --connect=udp:your_gcs_ip_address:14550**

your_gcs_ip_address is the address of the computer where the script is running 

I'll post also a more complex script with some commands like takeoff , starting a mission and so on

Ciao 

/Luca

```
#!/usr/bin/env python
# -*- coding: utf-8 -*-

from dronekit import connect, VehicleMode
import time
import argparse
import datetime

#Set up option parsing to get connection string                                                                                                                                                                   $parser = argparse.ArgumentParser(description='Print out vehicle state information. Connects to SITL on local PC by default.')
parser.add_argument('--connect',
                   help="vehicle connection target string. If not specified, exit.")
parser.add_argument('--uavid',
                   help="vehicle id. If not specified, exit.")
args = parser.parse_args()

connection_string = args.connect
connection_drone_id = args.uavid
sitl = None

#Start SITL if no connection string specified
if not connection_string or not connection_drone_id:
        print("Pls specify the connect parameters (--connect)")
        exit()

# Connect to the Vehicle.
#   Set `wait_ready=True` to ensure default attributes are populated before `connect()` returns.
print("\nConnecting to vehicle on: %s" % connection_string)
vehicle = connect(connection_string, wait_ready=True)

cmds = vehicle.commands
cmds.download()
cmds.wait_ready()

vehicle.wait_ready('autopilot_version')

##########
# Start
##########
while True:
        print vehicle.battery.level
        print vehicle.system_status.state
        print vehicle.armed
        print vehicle.heading
        print vehicle.groundspeed
        print vehicle.gps_0.fix_type
        print vehicle.gps_0.satellites_visible
        print vehicle.location.global_frame.lat
        print vehicle.location.global_frame.lon
        print vehicle.location.global_frame.alt
        print vehicle.heading
        print vehicle.groundspeed
        print vehicle.velocity[2]
        print vehicle.home_location.lat
        print vehicle.home_location.lon
        print vehicle.home_location.alt
        print vehicle.commands.next
        print vehicle.commands.count
```

The1only, 2020-12-17:
Perfekt, I work round the clock to get the mission planning working. The conversion from Mavlink to DJI and back is almost ready.

danammeansbear, 2020-12-17:
This is awesome 👌

adpmdrones, 2020-12-18:
> Perfekt, I work round the clock to get the mission planning working. The conversion from Mavlink to DJI and back is almost ready.

Awesome!!!

lukelu0520, 2020-12-18:
We are stuck in change mode from Stabilize to GUIDED. (by using the command `vehicle.mode = VehicleMode("GUIDED")`)
But it doesn't work to our DJI mini.

To our knowledge, drone can not be armable without going to GUIDED mode. 
Is there anyway to set the mode to GUIDED, maybe inside rosettadrone ?

Hope to hear your expertise.

Best,
-- Luke

The1only, 2020-12-18:
If I remember right the DJI drone switches automatically on a takeoff command from mavlink, ignore the error and continue to takeoff. But RosettaDrone expect 'P' mode on the controller (Mavic 2 Pro), and you must click on the "lock" so it becomes white NOT geeen before issuing the command.

lukelu0520, 2020-12-18:
Thanks for your persistently kind help, @The1only . By ignore mode chage, armable, arm check, we are able to takeoff ! 
To help more newbie folks, we paste the simple takeoff script below:

===
```
#Ignore Mode / Arms, directly takeoff to targetAltitude. 
#Advice: better to remove drone propellers before executing the script !
```

```
from dronekit import connect
import time
vehicle = connect('udpin:hostip:port', wait_ready=True)
vehicle.wait_ready('autopilot_version')
targetAltitude = 1.5
vehicle.simple_takeoff(targetAltitude)
while True:
    print(" Altitude: ", vehicle.location.global_relative_frame.alt)
    if vehicle.location.global_relative_frame.alt>=aTargetAltitude*0.95:
        print "Reached target altitude"
        break
    time.sleep(1)
```

The1only, 2020-12-19:
PS: remember that if you hit the SIM button on startup you can take off and fly like this without actually taking off. The SIM button enables the built inn DJI HITL module.

lukelu0520, 2020-12-19:
Nice to know the feature !
BTW, was wondering if you know how to construct a mavlink msg for LANDING ? So far we know it may requres:
mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT_INT, mavutil.mavlink.MAV_CMD_NAV_LAND
inside the mavlink packet.

The1only, 2020-12-19:
def return_home_and_land(self):
        # Send Drone home and land...        
        print("Waiting for drone to enter RTL flight mode")
        self.vehicle.mode = VehicleMode("RTL")
        time.sleep(5)
        while not self.vehicle.mode.name == 'RTL':
            time.sleep(0.5)

        print("Waiting for drone to return to landing")
        while self.vehicle.armed is True:
            print("Current Altitude: %f" % self.vehicle.location.global_relative_frame.alt)
            time.sleep(1.0)
        
        print("Landed...")

    def land(self):
        print("Now let's land")
        self.vehicle.mode = VehicleMode("LAND")
        time.sleep(5)

        print("Waiting for landing")
        while self.vehicle.mode == 'LAND' and self.vehicle.armed is True:
            print("Current Altitude: %f" % self.vehicle.location.global_relative_frame.alt)
            time.sleep(1)

        print("Landed...")

The1only, 2020-12-19:
In QGC you can also point and click on the location you want the drone to "goto" and it will.

lukelu0520, 2020-12-19:
The problem we met is set VehicleMode doesn't take effect. We've tested multiple VehicleMode such as GUIDED, LAND, but none of them is successfully set.

Alternatively, vehicle.send_mavlink(msg) works.

So we were wondering if there is another way to command LAND.

Best,
-- Luke

The1only, 2020-12-19:
Hmm strange I have been using this code for over one year now without issues. Let me look into it.

lukelu0520, 2020-12-19:
Thanks for your prompt reply. 
In the meantime, we will also log the behavior inside doInBackground of MainActivity.java.
Hope to print out msg_cmd.command for the analysis.

The1only, 2020-12-19:
It's allready there, just select the LOGCAT windows and enter MAV in the text window and select Verbose...
You will the see all you're commands.

lukelu0520, 2020-12-19:
We found that MAV_CMD_DO_SET_MODE (176) doesn't take effectiveness.
The mavlink msg is:

`sysid:255 compid:0 param1:1.0 param2:9.0 param3:0.0 param4:0.0 param5:0.0 param6:0.0 param7:0.0 command:176 target_system:1 target_component:0 confirmation:0`

And Rosetta Drone execute: `changeFlightMode((int) msg_cmd.param1);` 

Should we use msg_cmd.param1 or we should use msg_cmd.param2 ?   msg_cmd.param2 is 9, which seems matches LAND code better.

The1only, 2020-12-19:
Could you provide a short test case that creates this problem?
As you know there are two different MAVlink commands for changing the mode, both end up doing the same. 
Maybe there is something there. The long command 176 uses parameter 1, the command 11 uses parameter 2.
**https://mavlink.io/en/messages/common.html**

lukelu0520, 2020-12-19:
The python code of dronekit is below. VehicleMode("LAND") send command 176 to RosettaDrone, and  RosettaDrone uses command 176 to do the case MAV_CMD_DO_SET_MODE, and then changeFlightMode((int) msg_cmd.param2) needs to use to set proper mode for RosettaDrone.

The long command is: `sysid:255 compid:0 param1:1.0 param2:9.0 param3:0.0 param4:0.0 param5:0.0 param6:0.0 param7:0.0 command:176 target_system:1 target_component:0 confirmation:0`

Not sure why LAND parameter (9.0) is not put in param 1. It's put in 2 for Custom Mode. 

```
    def land(self):
        print("Now let's land")
        self.vehicle.mode = VehicleMode("LAND") 
        time.sleep(5)

        print("Waiting for landing")
        while self.vehicle.mode == 'LAND' and self.vehicle.armed is True:
            print("Current Altitude: %f" % self.vehicle.location.global_relative_frame.alt)
            time.sleep(1)

        print("Landed...")
```

The1only, 2020-12-19:
You are on to something, thank you.

On my MAC with all the latest modules, I get the same issue as you, that is, the Mode change command uses parameter 2, as it should. When I change the code to use parameter 2 rather than 1 then all the mode changes work again and I can take off and land.
However on my main Linux system that pulled modules 6 months ago. I need to use parameter 1.
I will check the DroneKit issue list, they may have fixed this.

Greate catch...

The1only, 2020-12-19:
I used a setup from a Udemy class years ago...  
I will check-in the fix and update all my systems.

The1only, 2020-12-19:
If you check out the “Work” branch it is fixed. 
I now uses Parameter 2 and can confirm it works on my MAC with all the latest.
But now it’s a problem on older systems.

lukelu0520, 2020-12-20:
Thanks for your fix, @The1only ! 

We are working on velocity control, regards to your example "Send a velocity command with +x being the heading of the drone." on the top of the discussion thread:

What is `int(x10000000), int(y10000000), float(z)` ?  Our python can not recognize x10000000 and y10000000.

And the line: `speed, speed, 1.0, # x, y, z velocity in m/s`, why you give 1.0 m/s for the z direction ?

Best,
-- Luke

adpmdrones, 2020-12-21:
We're experiencing the same issues:

from python script, we're able to takeoff and go to specified lat,lon

We've also an issue with goto command, the altitude will change to 1.2m

We're unable to issue command like RTL or LAND too

The1only, 2020-12-21:
Hmm We just fixed RTL did we not???
I installed everything fresh yesterday evening and it was working just fine.
It seems some DroneKit versions uses parameter 1 and some uses the correct parameter 2

Sorry for the bad example, I updated it now.

adpmdrones, 2020-12-21:
I must recompile Rosetta Drone?

The1only, 2020-12-21:
To get the latest fixes YES, we will not update master until Waypoint flying is working.

The1only, 2020-12-21:
OR you can just patch this:

                    case MAV_CMD_DO_SET_MODE:
                        changeFlightMode((int) msg_cmd.param2);
                        break;

The1only, 2020-12-21:
Ahh the GOTO command Altitude was also fixed some time back.
Maybe you should just upgrade to the "Work" patch.
Then you will also get DJI SDK 4.14 ( the brand new open)

adpmdrones, 2020-12-21:
Compiled with the work branch and SDK 4.14

Now RTH and LAND commands are ok
TAKEOFF it's OK
but the previous GOTO command (from python) does not work anymore

Still testing

BTW I cannot use GOTO from QGCS

The1only, 2020-12-21:
OK thank you I will fix.
Can you send the Python code you use for the GOTO?

adpmdrones, 2020-12-21:
# Arm and takeoff
arm_and_takeoff(7) <-- OK

time.sleep(5)

print("Going towards first point for 30 seconds ...")
vehicle.mode = VehicleMode("GUIDED")
point1 = LocationGlobalRelative(41.932030,  12.631848, 5)
point2 = LocationGlobalRelative(41.932327,  12.631982, 5)
vehicle.simple_goto(point1)
time.sleep(20)
vehicle.simple_goto(point2)
time.sleep(20)
print("Returning to Launch")
vehicle.mode = VehicleMode("RTL") <-- OK

The1only, 2020-12-21:
Perfect, never tested that command. Give me an hour...

adpmdrones, 2020-12-21:
Take your time, thank you

With the master branch, the GOTO command worked with the altitude issue

adpmdrones, 2020-12-21:
Let me know which command you use for guided

The1only, 2020-12-21:
OK This is part of a DJI issue that any mission MUST have two or more waypoints.
To fix this we use a trap, but somehow this fails now. I'll fix it.

I use another GOTO built from MAVlink Long command, but this is better so just way until I can test and verify my fix.

The1only, 2020-12-21:
I do not have a system to test on, but I belive I fixed the GOTO, and also added support for the old MissionItem/goto command that some still use...

adpmdrones, 2020-12-22:
Tested the updated work branch:

1 - simple_goto() dronekit function works as expected (altitude/heading ok)
2 - loading WPs (mission) from QGCS works as expected (mission)  (1)
3 - GOTO from QGCS works as expected (2)

Note:
1) After loading all the WPs, QGCS returns an error regarding the last ACK not correctly received but the mission was uploaded correctly
2) after sending from QGCS the GOTO command it returns an error but the command it's executed normally

adpmdrones, 2020-12-22:
I'll post some screenshots and python code later

adpmdrones, 2020-12-22:
[image]

that's the error after uploading waypoint (the mission works correctly)

The1only, 2020-12-22:
That is a known error I will fix it after I get the waypoint download to work (read back from the drone).

adpmdrones, 2020-12-22:
Thanks!!!

The same happens when using GOTO from QGCS

[image]

adpmdrones, 2020-12-23:
After some testing, we have reached a very reliable interaction between DJI and Mavlink (dronekit) scripts.

TakeOff, Landing, Brake, RTH, GOTO everything works fine from dronekit/python.
We need to test the mission loading and execution via dronekit/python.
We have also developed a REST API interface for getting the telemetry and send the commands to Rosetta Drone

More on this later

The1only, 2020-12-23:
Look forward to a rest api  example or two.Marry Christmas.

danammeansbear, 2020-12-23:
Can't wait for this!

adpmdrones, 2020-12-24:
Following an example of reading UAV data and publish them in a JSON format via HTTP:

#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
© Copyright 2019-2020, ADPM Drones.

Get  status from REST API (https://XXXXXXXXXXX)
vehicle status in JSON format returned:
{
  "status": {
    "status": "ACTIVE",
    "lastReachedWaypointId": 3,
    "name": "Drone ACME",
    "uavId": "1",
    "timestamp": "2020-12-13T07:18:34.867608+00:00",
    "EKF_status": true,
    "groundSpeedMetersPerSecond": 4.983015060424805,
    "is_armable": true,
    "heading": 274,
    "batteryRemainingPercent": 0,
    "lat": 99.729751,
    "alt": 591.88,
    "verticalSpeedMetersPerSecond": 0,
    "satellites": 10,
    "lng": 88.3460402,
    "armed": true,
    "fixType": 6,
    "channel": "LTE",
    "FlightMode": "AUTO"
  },
  "command": "HEARTBEAT"
}
"""
from __future__ import print_function
from dronekit import connect, VehicleMode
import time
import requests
import json
import datetime
import os
import pytz

"""
Set up option parsing to get connection string
"""
import argparse
parser = argparse.ArgumentParser(description="Read out vehicle state information and publich in JSON format.")
parser.add_argument("--connect",
                   help="vehicle connection target string.")
args = parser.parse_args()

connection_string = args.connect

"""
Connect to the Vehicle.
   Set `wait_ready=True` to ensure default attributes are populated before `connect()` returns.
"""
print("\nConnecting to vehicle on: %s" % connection_string)

while True:
	try:
		vehicle = connect(connection_string, wait_ready=True)
	except:
		print ("Some Dronekit error")
		"""
		Retry connection
		"""
		continue
	# if everything ok proceed
	break

vehicle.wait_ready("autopilot_version")

"""
Get all vehicle attributes (state)
"""
print("\nGet all vehicle attribute values:")
print(" Global Location: %s" % vehicle.location.global_frame)
print(" Global Location (relative altitude): %s" % vehicle.location.global_relative_frame)
print(" Local Location: %s" % vehicle.location.local_frame)
print(" Attitude: %s" % vehicle.attitude)
print(" Velocity: %s" % vehicle.velocity)
print(" GPS: %s" % vehicle.gps_0)
print(" Battery: %s" % vehicle.battery)
print(" EKF OK?: %s" % vehicle.ekf_ok)
print(" Last Heartbeat: %s" % vehicle.last_heartbeat)
print(" Heading: %s" % vehicle.heading)
print(" Is Armable?: %s" % vehicle.is_armable)
print(" System status: %s" % vehicle.system_status.state)
print(" Groundspeed: %s" % vehicle.groundspeed)    # settable
print(" Airspeed: %s" % vehicle.airspeed)    # settable
print(" Mode: %s" % vehicle.mode.name)    # settable
print(" Armed: %s" % vehicle.armed)    # settable

drone_id = 1
channel = "LTE"
path = "/home/yyyyyyyyy/"

print("Vehicle tracking info process running...")

try:
	while True:
		d = datetime.datetime.now(tz=pytz.utc).isoformat()
		my_json = {
					"command": "HEARTBEAT",
					"status":
					{
						"name": "Drone ACME",
						"uavId": str(drone_id),
						"timestamp" : d,
						"channel" : channel,
						"status": vehicle.system_status.state,
						"is_armable": vehicle.is_armable,
						"armed": vehicle.armed,
						"batteryRemainingPercent": vehicle.battery.level,
						"fixType": vehicle.gps_0.fix_type,
						"satellites": vehicle.gps_0.satellites_visible,
						"lat": vehicle.location.global_frame.lat,
						"lng": vehicle.location.global_frame.lon,
						"alt": vehicle.location.global_frame.alt,
						"heading": vehicle.heading,
						"groundSpeedMetersPerSecond": vehicle.groundspeed,
						"verticalSpeedMetersPerSecond" : vehicle.velocity[2],
						"lastReachedWaypointId": vehicle.commands.next,
						"FlightMode": vehicle.mode.name,
						"EKF_status": vehicle.ekf_ok
					}
				}

		with open(path + "pos.json", "w") as json_file:
			json.dump(my_json, json_file, indent = 4)
		os.system("sudo mv " + path + "pos.json /var/www/html/xxxxxxxxx/index.html")
		#json_file.close()
		time.sleep(0.1)
except:
	print("An exception occurred")
	pass
"""
dummy data on exit
so it"s possible to check for no data coming from vehicle
"""
d = datetime.datetime.now(tz=pytz.utc).isoformat()
my_json = {
			"command": "HEARTBEAT",
			"status":
			{
				"name": "Drone ACME",
				"uavId": str(drone_id),
				"timestamp" : d,
				"channel" : channel,
				"status": None,
				"is_armable": None,
				"armed": None,
				"batteryRemainingPercent": None,
				"fixType": None,
				"satellites": None,
				"lat": None,
				"lng": None,
				"alt": None,
				"heading": None,
				"groundSpeedMetersPerSecond": None,
				"verticalSpeedMetersPerSecond" : None,
				"lastReachedWaypointId": None,
				"FlightMode": None,
				"EKF_status": None
			}
		}

with open(path + "pos.json", "w") as json_file:
	json.dump(my_json, json_file, indent = 4)
os.system("sudo mv " + path + "pos.json /var/www/html/xxxxxxxxxx/index.html")
json_file.close()

"""
Close vehicle object before exiting script
"""
print("\nClose vehicle object")
vehicle.close()

print("Completed")

adpmdrones, 2020-12-24:
For public webhook endpoints, I use http://www.webhook.site

Very powerful!!!!

The1only, 2020-12-24:
https://github.com/patrickelectric/mavlink2rest

adpmdrones, 2020-12-24:
> https://github.com/patrickelectric/mavlink2rest

Good catch!!!

Thanks!!!

adpmdrones, 2020-12-24:
Tested right now with Rosetta Drone (mav2rest)

[image]

The1only, 2020-12-24:
If it works, give the guy a feedback, and we get more attention also to our project.

adpmdrones, 2020-12-24:
> If it works, give the guy a feedback, and we get more attention also to our project.

Already done!

The1only, 2020-12-24:
In this tread they mention POST also, did you test that?

https://discuss.ardupilot.org/t/mavlink-to-rest-api/51660

The1only, 2020-12-24:
Ps: I have also tested the Mavproxy, it also works. Most interesting is the abillity to split Mavlink streams, and the command line GCS.

adpmdrones, 2020-12-25:
> Ps: I have also tested the Mavproxy, it also works. Moust interesting is the abillity to split Mavlink streams, and the command line GCS.

I used mavproxy too for splitting mavlink stream to qgcs and my REST API scripts

[24 more replies]

closed 2023-04-08 by kripper