RosettaDrone/rosettadrone #41: socket details for bidirectional communication

lukelu0520, 2020-12-17:
Dear rosettadrone friends,

The git is really nice, we've studies some core parts of these codes and use them very well from GCS side.

Our external GCS is able to get telemetry from the rosettadrone device (i.e. Android device installed rosettadrone). Now we are figuring out how to send telemetry to rosettadrone device in order to achieve bidirectional communication.

In MainActivity, there is a function "createTelemetrySocket", the default port is 14550 as the external GCS's receiving port.
But we were wondering what is the port and address setting (or what's the default value) to allow rosettadrone device to receive telemetry from GCS ? (is the core module "MAVLinkReceiver" ?)

Hope to hear your expertise and get some guides.

Best,
-- Luke

The1only, 2020-12-17:
You simply sends data back on the same socket you receive data, no need for an additional socket. RosettaDrone supports two independant connections, however if you can use one for receiving and one for sending I have never tested.Terje

The1only, 2020-12-17:
If you click on the config button up in the right corner, you can set the ip destination, and port, and the secondary port etc. All this is well tested.

lukelu0520, 2020-12-17:
@The1only , thanks for your support !  It's nice to learn the same socket would be used and no need for an additional socket.

Just one follow-up question:  any simple test method to verify that RosettaDrone successfully receive a test message from an external GCS (like 'Hello world !' or 'Roger, I got you !') ? 

-- Luke

The1only, 2020-12-17:
Also look in the exa.ples thread, there is an example in the end there.

lukelu0520, 2020-12-17:
Rosetta Drone uses ArduPilot, I assume it's APM rather than PX4 stack.
But it's based on APM 1 or APM 2?

The1only, 2020-12-17:
No there is not PX4 nor APM code in RosettaDrone, it's DJI flight controller. It uses the Mavlink 2 stack thats all, and the DJI sdk stack, but there is now flightcontroller in RosettaDrone...

closed 2020-12-17 by lukelu0520

MariusBeul, 2022-01-13:
@The1only
Sorry to reopen this issue. You state that Rosettadrone uses MAVLINK 2, but when I understand correctly, the [Parser.java](https://github.com/The1only/rosettadrone/blob/master/app/src/main/java/com/MAVLink/Parser.java) file [deserializes ](https://mavlink.io/en/guide/serialization.html) only MAVLINK 1 packages correctly. Also the Packet start marker is set to [0xFE](https://github.com/The1only/rosettadrone/blob/5bdcdce184a964e5fd5c9e6d75bc3f28b910bdda/app/src/main/java/com/MAVLink/MAVLinkPacket.java#L103) (254) which indicates MAVLINK 1.