RosettaDrone/rosettadrone #46: Seems we use Ardupilots non standard implementation of MavLink

The1only, 2020-12-30:
PX4 is fully compatible with Mavlink, however, we seem to use the Ardupilot version of mavlink. 
(Generated from the Mavlink generator tool).

What made me look more into this is the very first waypoint we receive from the GCS, it is a home location at 0 alt. wp.
This is a nonstandard Ardupilot way of doing things, we support both as we ignore this wp when it is at 0 alt ( makes no sense). However, we do use it if the next command is a takeoff (without position).

This is quite technical, but there is also an issue regarding wp position resolution. We need to look into this, as we need the best position accuracy we can get. I need to move exactly 1meters away, no rounding... 

If any of you knows something about this, please jump in.

adpmdrones, 2021-01-06:
If I understood well your post, pls check the following https://github.com/mavlink/qgroundcontrol/issues/6371

adpmdrones, 2021-01-06:
..and this

https://github.com/ArduPilot/ardupilot/issues/1379

kripper, 2023-02-17:
I have been playing with MAVSDK. They are also not compatible with ArduCopter SITL.
Missions can be sent but not started. Will fix it now.
Any way, I have been testing QGC with ArduCopter SITL to replicate the protocol on Rosetta.
In Rosetta 3, everyhting is working fine for me with ArduCopter SITL and DJI Mini SE.

closed 2023-02-17 by kripper