RosettaDrone/rosettadrone #54: Issues 'registering SDK' caused by 'check connection', how to resolve?

thomasgauvin, 2021-03-18:
**Describe the bug**
Encountering this issue on both current master and the release tag 4.11.1.2.  Toast messages show 'Disconnected', and 'Register sdk fails, check network is available', despite having connection.

**To Reproduce**
Steps to reproduce the behavior:
- Not sure if this is specific to my configuration, any help would be appreciated.

The1only, 2021-03-18:
This is coming from the DJI SDK, are you sure you entered the right username and password ????

thomasgauvin, 2021-03-18:
When I open the application, I do not receive a prompt for username and password. I have put my DJI key in /values/keys.xml however.

Also, I would note that I am trying this on Mavic Mini, although I believe it should work since it is supported my the DJI SDK

The1only, 2021-03-19:
If you do not get a prompt for username and password then uninstall Rosettadrone and reinstall it, you must log inn the first time.

Best regard
Terje Nilsen
9Tek AS
Norway

thomasgauvin, 2021-03-20:
It was an issue with my DJI SDK key or my Google key!

closed 2021-03-20 by thomasgauvin