RosettaDrone/rosettadrone #55: Can't build in Android Studio

DIOGOFIGUEIRA, 2021-03-23:
Ubuntu 18.04.5 LTS
Android Studio 4.1.2

I need some help with topics 3 and 4 of the "building from source" tutorial:
3 - what did you mean by "The package name should be sq.rogue.rosettadrone."?
4 - Where do I find the "/ values" folder?

In the Manifest.xml file, could you give me an example of how I replace the syntax: (android: value = "@ string / dji_key" />)
and (android: value = "@ string / google_key" />)?

Thanks.

gyroplast, 2021-03-23:
You do not edit the Manifest.xml directly. Create the file `app/src/main/res/values/keys.xml` instead, with this content:

```
<?xml version="1.0" encoding="utf-8"?>
<resources>
    <string name="dji_key">DJI_APP_KEY_OBTAINED_FROM_DJI</string>
    <string name="google_key">YOUR_GOOGLE_MAPS_API_KEY</string>
</resources>
```
You'll find this file's path already included in the main `.gitignore` to prevent your secrets from leaving your local workspace, and the Manifest.xml will pull in these values properly once they are defined. AndroidStudio will show you the referenced key values in the Manifest.xml if they are defined correctly in keys.xml.

The "package name" becomes relevant when you are requesting your personal DJI Application SDK key from the DJI Developer website. There'll be a field asking for this package name in the DJI request form. You'll have to register as a DJI developer (for free). Similar procedure for the Google Maps API key.

DIOGOFIGUEIRA, 2021-03-24:
gyroplast, first of all, thank you for the tips above.
I followed all the instructions and now received the following error:

Build command failed.
Error while executing process /root/Android/Sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /media/diogo/HD_1TB/Programas_Python/rosettadrone-master_v2/rosettadrone-master/app/.cxx/cmake/debug/x86 djivideojni}
ninja: Entering directory `/media/diogo/HD_1TB/Programas_Python/rosettadrone-master_v2/rosettadrone-master/app/.cxx/cmake/debug/x86'
[1/1] Linking C shared library ../../../../build/intermediates/cmake/debug/obj/x86/libdjivideojni.so
FAILED: ../../../../build/intermediates/cmake/debug/obj/x86/libdjivideojni.so 
: && /root/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=i686-none-linux-android22 --gcc-toolchain=/root/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/root/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -mstackrealign -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libdjivideojni.so -o ../../../../build/intermediates/cmake/debug/obj/x86/libdjivideojni.so CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o  ../../../../src/main/cpp/lib/x86/libffmpeg.so /root/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/i686-linux-android/22/liblog.so -latomic -lm && :
ld: error: ../../../../src/main/cpp/lib/x86/libffmpeg.so:1: unknown directive: version
>>> version https://git-lfs.github.com/spec/v1
>>> ^
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

Could you help me with that too?
Thanks.

The1only, 2021-03-24:
You are trying to build for Intel, but we only support Arm. The Intel sim version have been broken sinse day one. Would have been nice to have thow.

Best regard
Terje Nilsen
9Tek AS
Norway

DIOGOFIGUEIRA, 2021-03-24:
Hi Terje Nilsen,
Sorry for my lack of knowledge.
Just so I understand, should this program be compiled on a computer with RISC architecture, with an ARM processor? That's it?
I'm actually compiling from a computer with an INTEL processor (SISC architecture), but I have an Nvidia Jetson TX2.
Do you think I will be able to compile it?

Thank you very much for your help.

Diogo Figueira

The1only, 2021-03-24:
You can use any computer that run Android Studio, but must set the target to be for android arm, normally if you connect a cellphone or ann Android pad, it will automatically be selected as the target. If you have no device connected it will try to build for the Android emulator under the intel platform (as long as you are on a intel PC). The system will launch an android emulator, however Rosettadrone does not support the Android on Intel combination. Most of the code is there but was at some point abandoned. For a while I even removed the option in the manuskript but added it back, in case some one whant to look at it.

To run Rosettadrone on Nvidia Jetson, the Jetson must run Android 5.1 or newer. I have never tried it, I do got several Nvidia embedded boards but non running Android. If you get Jetson to run Android with full USB support... maybe.

Best regard
Terje Nilsen
9Tek AS
Norway

DIOGOFIGUEIRA, 2021-03-24:
Even though I added my Asus Next 7 tablet to Android Studio, the error remains the same ...
What can I be doing wrong?

Build command failed.
Error while executing process /root/Android/Sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /media/diogo/HD_1TB/Programas_Python/rosettadrone-master_v2/rosettadrone-master/app/.cxx/cmake/debug/armeabi-v7a djivideojni}
ninja: Entering directory `/media/diogo/HD_1TB/Programas_Python/rosettadrone-master_v2/rosettadrone-master/app/.cxx/cmake/debug/armeabi-v7a'
[1/1] Linking C shared library ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libdjivideojni.so
FAILED: ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libdjivideojni.so 
: && /root/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=armv7-none-linux-androideabi22 --gcc-toolchain=/root/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/root/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--exclude-libs,libunwind.a -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libdjivideojni.so -o ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libdjivideojni.so CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o  ../../../../src/main/cpp/lib/armeabi-v7a/libffmpeg.so /root/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/arm-linux-androideabi/22/liblog.so -latomic -lm && :
ld: error: ../../../../src/main/cpp/lib/armeabi-v7a/libffmpeg.so:1: unknown directive: version
>>> version https://git-lfs.github.com/spec/v1
>>> ^
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

closed 2021-03-25 by DIOGOFIGUEIRA

xzhub, 2021-07-01:
@DIOGOFIGUEIRA have you resolved this issue? I am seeing the same message:
[image]
[image]

I believe I entered the correct information to generate the DJI app key:
[image]

@The1only Would you please take a look and let me know how to resolve this issue?
BTW, why you don't push this app into google play so we can download and install it directly?:)

The1only, 2021-07-01:
Did you registrer correctly?  This is not a common prpblem.

Best regard
Terje Nilsen
9Tek AS
Norway

xzhub, 2021-07-01:
@The1only , This is hard to answer because I don't know if there is any way to test my key. For example, if there is any curl command I can use to test if the key is working or not? I only know that I can login DJI web site with no issue and I registered that android app with the following information:

APP Name: Rosetta Drone 2
Software Platform: Android
Bundle Identifier: sq.rogue.rosettadrone
Category: AI
Description: DJI MAVLINK WRAPPER

I've shared the screenshot of this registered app in my previous post.

xzhub, 2021-07-02:
I deleted the previous app key, then created another one on the DJI website https://developer.dji.com/. But unfortunately, the error is the same...

xzhub, 2021-07-02:
Error message from Android Studio:

```
07/01 21:59:17: Launching 'app' on LGE Nexus 5.
Install successfully finished in 1 m 3 s 135 ms.
$ adb shell am start -n "sq.rogue.rosettadrone/sq.rogue.rosettadrone.ConnectionActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 29120 on device 'lge-nexus_5-0ad9c0b2d048c330'.
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/art: Late-enabling -Xcheck:jni
D/sq.rogue.rosettadrone.MainActivity: checkAndRequestPermissions
D/SDKRelativeJNI: load lib suc
D/TAG: onInitProcess
D/FPVController: try to load libdjivideo.so
V/sq.rogue.rosettadrone.MainActivity: Version: 5.1.1
V/sq.rogue.rosettadrone.MainActivity: onResume
D/NativeRcController: x try to load libusbdec.so
D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: true
E/NativeRcController: Couldn't load libusbdec.so
I/Adreno-EGL: <qeglDrvAPI_eglInitialize:379>: QUALCOMM Build: 10/21/15, 369a2ea, I96aee987eb
I/OpenGLRenderer: Initialized EGL, version 1.4
I/onLoad: loaded onLoad.cpp :JNI_OnLoad() go to end
D/GroudStation: load lib suc
D/DJIFlycParamInfoManager: readToMemory size = 705
V/RenderScript: 0x95675000 Launching thread(s), CPUs 4
E/DJIUsbAccessoryReceiver: start accessory receiver
D/UsbAccessoryService: accessory startStream
E/DJIUsbAccessoryReceiver: receive action: android.hardware.usb.action.USB_STATE
E/DJIUsbAccessoryReceiver: android.hardware.usb.action.USB_STATE: is connected
D/logWrite.cpp: UDT JNI_OnLoad---------------->
    UDP jniRegisterNativeMethods: jniRegisterNativeMethods
    loaded UDT :JNI_OnLoad() go to end
E/Lightbridge: Failed to obtain JNIEnv
E/Lightbridge: Failed to obtain JNIEnv
E/Lightbridge: Failed to obtain JNIEnv
E/Lightbridge: Failed to obtain JNIEnv
E/Lightbridge: Failed to obtain JNIEnv
E/Lightbridge: Failed to obtain JNIEnv
E/RM500 RCType: Service Not Connected
E/PM430 RCType: Service Not Connected
D/HttpsHelper: verified, return
A/libc: Fatal signal 11 (SIGSEGV), code 2, fault addr 0x922b2000 in tid 29204 (UrgentThreadPoo)
```

xzhub, 2021-07-02:
I guess that Nexus 5 is too old (Android 5.x) to run this app. After I switched to Google Pixel with Android OS 10, the registration succeeded.

xzhub, 2021-07-02:
@The1only I'd like to mention that DJI GO 4 works perfectly on my Nexus 5, there is no issue at all. But ResettaDrone2 can't finish the registering process.

The1only, 2021-07-02:
I belive we had to abandon 5.1x support when we added google maps… you can try reenable 5.1 support.

xzhub, 2021-07-02:
Current status update: with the latest code from work branch (2d10989), I can run the app on my Google Nexus 5 and connect to my Mavic Pro via WiFi. The issue is that the SDK is not stable, still can't use it yet.
[image]

xzhub, 2021-07-03:
Strange enough that now GO4 is working on my Pixel and RosettaDrone2 could connect to Mavic as well.
Currently, RosettaDrone2 work branch works on My Nexus 5/Pixel/Pixel 3XL, all show the same msg: "Oops,DJI's SDK just glitched. Please restart the app.". The debug log on Pixel 3XL is pasted below. @The1only could you please take look and see what might be the cause? I'd like to see if there is any settings I can chagne then try the app again.

One error msg I noticed in all 3 phones is:

```
I/Choreographer: Skipped 51 frames!  The application may be doing too much work on its main thread.

```

The full log is attached.
[Pixel3XL.log](https://github.com/The1only/rosettadrone/files/6757798/Pixel3XL.log)

The1only, 2021-07-03:
The phone is too slow, and must dropp frames.

Best regard
Terje Nilsen
9Tek AS
Norway

xzhub, 2021-07-06:
I see the same issue with all my 3 phones, from Nexus 5, to Pixel and Pixel 3XL(I think this one should be fast enough). All of them works fine with cable connection to the controller, but report the skipped frames error msg with WiFi connection to drone. Seems like an issue with WiFi only.

fajrulfx, 2021-07-08:
> gyroplast, first of all, thank you for the tips above.
> I followed all the instructions and now received the following error:
>
> Build command failed.
> Error while executing process [...]

@DIOGOFIGUEIRA can you please share how you solve this issue? I'm stuck in this problem although I already build for ARM only (via build.gradle setting).

xzhub, 2021-07-10:
@fajrulfx If you saw this error while compiling the downloaded release zip file, you may want to delete the folder and use git to clone the latest master branch or even the work branch of the code. After you create the keys.xml, you can start to build the project. I think I saw this error while building the code from release zip file.

The1only, 2021-07-10:
This error normally happens when GIT LFS is not installed when you check out the files from git.

Best regard
Terje Nilsen
9Tek AS
Norway

fajrulfx, 2021-07-12:
Thank you @The1only and @xzhub for the advice. I've successfully build the project on Android Studio by following these steps:
1. Install git lfs on my Mac `brew install git lfs` and activate it using `git lfs install`
2. Delete the old project folder and use git lfs to clone repository `git lfs clone https://github.com/The1only/rosettadrone`
3. Build > Make Project on Android Studio

xzhub, 2021-07-12:
@fajrulfx Good to know it works! Does the WiFi connection works on your phone?

DIOGOFIGUEIRA, 2021-07-12:
xzhub 

I had this issue on google Nexus 7 smartphone.
After that I tried it with a Samsung Galaxy smartphone and it worked.