RosettaDrone/rosettadrone #56: Got the 5/4G LTE flying to work today using VPN.

The1only, 2021-03-23:
This is super cool, by using a Mavic 2 Pro connected to the origional controller, and it connected to an 4G Android phone that used OpenVPN to connect to a VPN server we set ut, then using a Mac on the same VPN, from a totally different location, that ran Qgroundcontroll. We was able to demonstrate live video with quite low latency (less that 1second), and full bidirectional command and controll. Using a Joystik or mission we can fly a DJI drone remotly from any location on the planet that got internet... this opens quite a few cool applications. 
Just to let you know!!!!

pierotofy, 2021-04-05:
That's quite cool!

xzhub, 2021-07-02:
@The1only Could we connect the android phone directly to Mavic pro via wifi without the controller? I'd like to mount the phone to the DJI drone, then use Dronekit/QGC to control the drone over 4G network. I would be super excited if you can add that feature! Currently, Rosetta Drone can't find my drone if I connect the phone to the drone vai WIFI.

The1only, 2021-07-02:
Hmm I tought I tested that with the old Air, there are some limitations that will applay, but in general it should work. Maybe I remember wrong. Do GO4 work?

xzhub, 2021-07-02:
Just installed GO4 on Nexus 5 and it worked fine.
[image]

xzhub, 2021-07-02:
Ooops. we are chatting in the wrong thread.

xzhub, 2021-07-02:
No!!  GO4 does not work either on my Google Pixel with Android 10. No way!

kripper, 2023-02-21:
Share videos. It's ok if you advertise your company.

closed 2023-02-21 by kripper