RosettaDrone/rosettadrone #58: Override guided point does not work

adpmdrones, 2021-03-26:
Hi,
I'm attempting to send guided point B to drone while it is still in guided but toward previously sent guided point A. Unfortunately drone continues to go toward guided point A. I attempted to set "brake" mode, and then to provide guided point B, but nothing happens (rather, in this situation drone can't move at all, neither to point A if sent again).
I tried using Mission Planner simulator instead of Rosetta/DJI and it works.
Is this a DJI SDK limitation?
Thank you in advance

The1only, 2021-03-26:
I belive you are ahead of development, feel free to dig inn.

Best regard
Terje Nilsen
9Tek AS
Norway

lilfish, 2021-04-20:
> Is this a DJI SDK limitation?

No, it is not yet programmed I believe.

kripper, 2023-08-15:
Check latest version.

closed 2023-08-15 by kripper