RosettaDrone/rosettadrone #59: Support for Mavic air 2

zhouzhiwen2000, 2021-03-27:
Is there plan to support mavic air 2? The sdk supporting it is released recently.

zhouzhiwen2000, 2021-03-27:
I built and tested the current source, there is no video, other things are functioning.

The1only, 2021-03-27:
One of us that work on the video needs to get our hands on a unit. There are probably small modification, like adding the drone to the right path in the code.

Best regard
Terje Nilsen
9Tek AS
Norway

zhouzhiwen2000, 2021-03-28:
Looking forward to it!

zhouzhiwen2000, 2021-03-31:
Seem like the problem is with decoding.
Error msg:
E/NativeHelper: NativeHelper
A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x148 in tid 29216 (parseVideoFromC), pid 29034 (ue.rosettadrone)

The1only, 2021-03-31:
Try add the drone ID to the list of Ocusync drones, like the Mavic etc.

Best regard
Terje Nilsen
9Tek AS
Norway

zhouzhiwen2000, 2021-03-31:
is the list validateTranscodingMethod?
Added, no difference.
Also, the video view on the phone doesn't work.
The video view on the phone uses DJI decoder by calling sendDataToDecoder, so it shouldn't be affected, right?
The same decoder works find on the Sample code.

The1only, 2021-03-31:
I seen that before, there is a call back function that calls the DJI decoder, it ofthen has to be adapted. On some systems we get pure H264, on other it is DJI version of the H264 that need transcoding. I do not have the code with me, but look for the video frame call back function in the Main code and play with the different paths... 
You are on to something, first get the App screen video to work, then focus on the export.
Ones you got it to work we will clean it up.

Best regard
Terje Nilsen
9Tek AS
Norway

zhouzhiwen2000, 2021-04-03:
The video works simply by adding MAVIC_AIR_2 to validateTranscodingMethod.
The problem is with my phone.
With this added, video can be sent via UDP.
The display on the phone seems to be a phone-specific problem.
I am using Huawei Mate 20 X 5G, running Android 10.
Please test the app on devices with Android 10.

zhouzhiwen2000, 2021-04-03:
> The video works simply by adding MAVIC_AIR_2 to validateTranscodingMethod.
> The problem is with my phone.
> With this added, video can be sent via UDP.
> The display on the phone seems to be a [...]

I am extremely sorry, I updated the Gradle, and that caused all the problems.
Add MAVIC_AIR_2 to validateTranscodingMethod, and it'll work flawlessly.

The1only, 2021-04-03:
Yes, that is known, at some point, we should figure that out. Glad you got it, I'll add the fix, or you can create a push if you want.

zhouzhiwen2000, 2021-04-04:
OK, created a PR.

closed 2021-04-04 by zhouzhiwen2000