RosettaDrone/rosettadrone #61: Flight control and video capture via python

DIOGOFIGUEIRA, 2021-03-29:
I'm trying to control the drone's flight and also capture the video via python, using Qgroundcontron on Ubuntu.

My phantom 4 pro did not connect with the Rosetta Drone.
So, I'm thinking of buying another drone to be able to use this release of the rosetta drone app.
At this point, which drone would I be able to achieve these goals (flight control and video capture)?

You commented that both Mavic Air 2 and Mavic Air 2 Pro would probably work.
However, there is a considerable difference in price between them.
Which drone model do you feel most comfortable using the rosetta drone at this moment?

Thank you for your consideration.

Diogo Figueira

The1only, 2021-03-29:
The Mavic 2 Pro/Zoom/Enterprice is what was used making the code, so it’s the safest beth.

Best regard
Terje Nilsen
9Tek AS
Norway

lilfish, 2021-04-17:
Phantom 4 is not enabled in the code but has been tested in previous tests. 
Enable the phantom 4 by adding it to the list in [MainActivity.java](https://github.com/The1only/rosettadrone/blob/master/app/src/main/java/sq/rogue/rosettadrone/MainActivity.java)

Adding the following cases:

```java
case PHANTOM_4_STANDARD:
case PHANTOM_4_ADVANCED:
case PHANTOM_4_PROFESSIONAL:
```
worked, but not sure which one did the trick.

DIOGOFIGUEIRA, 2021-04-19:
I added the lines to the MainActivity.java file.
The messages say that it has been registered and that it has been connected with the Phantom 4 Pro.
Then I hit the OPEN button and it turns gray.
After that, should another screen appear?

I open Qgrouncontrol, but it says that it has lost connection with the drone.

Thank you for your help

lilfish, 2021-04-21:
After the button is pressed you should be welcomed with the application view, could you maybe share your logs? This can be done using catlog.

Thanks in advance

DIOGOFIGUEIRA, 2021-05-13:
Hello lilfish,
I replaced my phantom 4 pro with a mavic 2 pro,
but the error remains the same.
Below, the log you requested.
Thank you very much for your help so far.

build.board: MSM8226
build.bootloader: 0x411A
build.brand: motorola
build.cpu_abi: armeabi-v7a
build.cpu_abi2: armeabi
build.device: peregrine
build.display: LPBS23.13-35.5-1
build.fingerprint: motorola/peregrine_retbr/peregrine:5.1/LPBS23.13-35.5-1/1:user/release-keys
build.hardware: qcom
build.host: ilclbld52
build.id: LPBS23.13-35.5-1
build.manufacturer: motorola
build.model: XT1040
build.product: peregrine_retbr
build.radio: unknown
build.serial: 0433699103
build.tags: release-keys
build.time: 1460542609000
build.type: user
build.user: hudsoncm
version.codename: REL
version.incremental: 1
version.release: 5.1
version.sdk_int: 22

--------- beginning of main
05-12 22:36:26.028 W/m       (10214): Cannot obtain root
05-12 22:36:26.028 W/m       (10214): java.io.IOException: Error running exec(). Command: [su] Working Directory: null Environment: null
05-12 22:36:26.028 W/m       (10214): at java.lang.ProcessManager.exec(ProcessManager.java:211)
05-12 22:36:26.028 W/m       (10214): at java.lang.Runtime.exec(Runtime.java:173)
05-12 22:36:26.028 W/m       (10214): at java.lang.Runtime.exec(Runtime.java:246)
05-12 22:36:26.028 W/m       (10214): at java.lang.Runtime.exec(Runtime.java:189)
05-12 22:36:26.028 W/m       (10214): at com.nolanlawson.logcat.b.m.a(Unknown Source)
05-12 22:36:26.028 W/m       (10214): at com.nolanlawson.logcat.b.w.a(Unknown Source)
05-12 22:36:26.028 W/m       (10214): at com.nolanlawson.logcat.b.p.b(Unknown Source)
05-12 22:36:26.028 W/m       (10214): at com.nolanlawson.logcat.d.doInBackground(Unknown Source)
05-12 22:36:26.028 W/m       (10214): at android.os.AsyncTask$2.call(AsyncTask.java:292)
05-12 22:36:26.028 W/m       (10214): at java.util.concurrent.FutureTask.run(FutureTask.java:237)
05-12 22:36:26.028 W/m       (10214): at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
05-12 22:36:26.028 W/m       (10214): at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
05-12 22:36:26.028 W/m       (10214): at java.lang.Thread.run(Thread.java:818)
05-12 22:36:26.028 W/m       (10214): Caused by: java.io.IOException: Permission denied
05-12 22:36:26.028 W/m       (10214): at java.lang.ProcessManager.exec(Native Method)
05-12 22:36:26.028 W/m       (10214): at java.lang.ProcessManager.exec(ProcessManager.java:209)
05-12 22:36:26.028 W/m       (10214): ... 12 more
--------- beginning of system
05-12 22:37:15.686 W/ResourcesManager(10214): Asset path '/system/framework/com.google.android.maps.jar' does not exist or contains no resources.
05-12 22:37:15.842 W/ResourcesManager(10214): Asset path '/system/framework/com.google.android.media.effects.jar' does not exist or contains no resources.
05-12 22:37:15.911 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:15.911 W/PackageManager(10214): Failure retrieving resources for com.android.bluetooth: Resource ID #0x0
05-12 22:37:15.927 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:15.928 W/PackageManager(10214): Failure retrieving resources for com.alphainventor.filemanager: Resource ID #0x0
05-12 22:37:15.943 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:15.944 W/PackageManager(10214): Failure retrieving resources for com.google.android.gm: Resource ID #0x0
05-12 22:37:15.960 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:15.960 W/PackageManager(10214): Failure retrieving resources for com.google.android.apps.plus: Resource ID #0x0
05-12 22:37:15.973 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:15.974 W/PackageManager(10214): Failure retrieving resources for com.google.android.talk: Resource ID #0x0
05-12 22:37:15.992 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:15.992 W/PackageManager(10214): Failure retrieving resources for com.android.mms: Resource ID #0x0
05-12 22:37:15.996 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:15.997 W/PackageManager(10214): Failure retrieving resources for com.google.android.apps.docs: Resource ID #0x0
05-12 22:37:16.023 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:16.023 W/PackageManager(10214): Failure retrieving resources for com.android.bluetooth: Resource ID #0x0
[... 57 lines cut ...]
05-12 22:37:16.436 W/PackageManager(10214): Failure retrieving resources for com.google.android.talk: Resource ID #0x0
05-12 22:37:16.439 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:16.440 W/PackageManager(10214): Failure retrieving resources for com.android.mms: Resource ID #0x0
05-12 22:37:16.442 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:16.442 W/PackageManager(10214): Failure retrieving resources for com.google.android.apps.docs: Resource ID #0x0
05-12 22:37:16.451 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
[... cut ...]
05-12 22:37:21.885 W/PackageManager(10214): Failure retrieving resources for com.google.android.apps.docs: Resource ID #0x0
05-12 22:37:22.009 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:22.009 W/PackageManager(10214): Failure retrieving resources for com.google.android.apps.translate: Resource ID #0x0
05-12 22:37:24.851 W/ResourceType(10214): No package identifier when getting value for resource number 0x00000000
05-12 22:37:24.851 W/PackageManager(10214): Failure retrieving resources for com.android.bluetooth: Resource ID #0x0

cepoya, 2021-07-21:
> The Mavic 2 Pro/Zoom/Enterprice is what was used making the code, so it’s the safest beth. Best regard Terje Nilsen 9Tek AS Norway
> […](#)

actually i used mavic 2, i need get raw data telemetry and video trough to my computer (in same network) with Pycharm (python), can you give the solusion?

closed 2021-07-21 by DIOGOFIGUEIRA