RosettaDrone/rosettadrone #66: Add more documentation

lilfish, 2021-04-17:
**Is your feature request related to a problem? Please describe.**

Add more code documentation for contributors

**Describe the solution you'd like**
Either use the wiki or a docs folder with different markdown files containing documentation on parts of the code.

**Describe alternatives you've considered**
docs folder or wiki could be used. Preferable docs as this will be included in one repository rather than two.

Docs for the different java classes as well as contributing documentation if possible

lilfish, 2021-04-17:
Example document on FTP:
https://github.com/lilfish/Delta-FlowerPower/wiki/Droneapp-Python-FTP

(May be fully copied)

kripper, 2023-02-17:
Thanks, we added your doc in the README page.

closed 2023-02-17 by kripper