RosettaDrone/rosettadrone #67: Compilation errors

highfreq, 2021-05-01:
Hello and thank you for this great project.

I am very very very new to android studio.

Version Android Studio 4.1.3

Samsung S21 connected to my pc

PC operating system Ubuntu 20.04.2 LTS

Error:

Build command failed.
Error while executing process /home/corrado/Android/Sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /home/corrado/rosettadrone-master/app/.cxx/cmake/debug/armeabi-v7a djivideojni}
ninja: Entering directory `/home/corrado/rosettadrone-master/app/.cxx/cmake/debug/armeabi-v7a'
[1/2] Building C object CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o
In file included from ../../../../src/main/cpp/dji_video_jni.c:4:
In file included from ../../../../src/main/cpp/ffmpeg.h:34:
In file included from ../../../../src/main/cpp/cmdutils.h:28:
In file included from ../../../../src/main/cpp/include/libavcodec/avcodec.h:31:
In file included from ../../../../src/main/cpp/include/libavutil/samplefmt.h:24:
In file included from ../../../../src/main/cpp/include/libavutil/avutil.h:289:
In file included from ../../../../src/main/cpp/include/libavutil/common.h:401:
In file included from ../../../../src/main/cpp/include/libavutil/mem.h:33:
../../../../src/main/cpp/include/libavutil/error.h:121:5: warning: non-constant static local variable in inline function may be different in different files [-Wstatic-local-in-inline]
    static char str[AV_ERROR_MAX_STRING_SIZE];
    ^
../../../../src/main/cpp/include/libavutil/error.h:119:1: note: use 'static' to give inline function 'av_err2str' internal linkage
av_always_inline char *av_err2str(int errnum) {
^
static 
../../../../src/main/cpp/include/libavutil/attributes.h:37:30: note: expanded from macro 'av_always_inline'
#    define av_always_inline __attribute__((always_inline)) inline
                             ^
1 warning generated.
[2/2] Linking C shared library ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libdjivideojni.so
FAILED: ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libdjivideojni.so 
: && /home/corrado/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=armv7-none-linux-androideabi22 --gcc-toolchain=/home/corrado/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/corrado/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--exclude-libs,libunwind.a -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libdjivideojni.so -o ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libdjivideojni.so CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o  ../../../../src/main/cpp/lib/armeabi-v7a/libffmpeg.so /home/corrado/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/arm-linux-androideabi/22/liblog.so -latomic -lm && :
ld: error: ../../../../src/main/cpp/lib/armeabi-v7a/libffmpeg.so:1: unknown directive: version
>>> version https://git-lfs.github.com/spec/v1
>>> ^
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

highfreq, 2021-05-01:
For others that may encounter probs compiling this is what resolved my prob:

Go to https://github.com/git-lfs/git-lfs/wiki/Installation and install git fls

on terminal run:

git lfs install

It should say "Git LFS initialized." 

Then clone the repository

GIT_SSL_NO_VERIFY=1 git clone https://github.com/The1only/rosettadrone.git

Open cloned repository with android studio and builld :)

VladimirPobedinskiy, 2021-05-18:
I'm install lfs but:

```[2/2] Linking C shared library libdjivideojni.so
FAILED: libdjivideojni.so 
cmd.exe /C "cd . && C:\Users\cin\AppData\Local\Android\Sdk\ndk\22.0.7026061\toolchains\llvm\prebuilt\windows-x86_64\bin\clang.exe --target=armv7-none-linux-androideabi22 --gcc-toolchain=C:/Users/cin/AppData/Local/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/cin/AppData/Local/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/windows-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--exclude-libs,libunwind.a -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libdjivideojni.so -o libdjivideojni.so CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o  ../../../../src/main/cpp/lib/armeabi-v7a/libffmpeg.so -llog -latomic -lm && cd ."
ld: error: ../../../../src/main/cpp/lib/armeabi-v7a/libffmpeg.so:1: unknown directive: version
>>> version https://git-lfs.github.com/spec/v1
>>> ^
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

closed 2023-02-24 by kripper