RosettaDrone/rosettadrone #68: Dronekit connection

gab933, 2021-05-06:
Hi,
I have connection issue with Rosetta and Dronekit. When I launch first time Rosetta and then I connect dronekit everything works. But if I disconnect dronekit, and then I attempt to connect it again (by leaving Rosetta running), I incur in connection timeout. Therefore I have to close Rosetta and launching it again.
How can I solve this problem?
Thank you in advance

The1only, 2021-05-06:
Do you receive status but can’t send any commands? Or do you get nothing?

Rosettadrone sends data blindly out on port 14550 (default) and knows nothing about the receiving side or even if any one is listhning.

Can’t say I have ever seen this problem, and I use Dronekit all the time.

Are you shure you exit Dronkit propperly, with closed UDP port.

Best regard
Terje Nilsen
9Tek AS
Norway

gab933, 2021-05-10:
Thank you for your reply.
I have problems with re-connecting dronekit by leaving Rosetta up. I close dronekit in the correct way by using vehicle.close() function. When I stop and re-open Rosetta, connection works as expected, so it looks like Rosetta toils to re-establish connection with dronekit. I performed a simulation with Arducopter Mission Planner (as source instead of Rosetta) and I successfully close and re-open dronekit connection lot of times without re-launching it.
Best regards,

Gabriele

The1only, 2021-05-11:
Hmmm, I do this all the time, I will check my code, maybe there is a trick I have forgotten. 
I really do not see how this can be.

Best regard
Terje Nilsen
9Tek AS
Norway

gab933, 2021-05-11:
Hi Terje,
an update: I noticed that by setting wait_ready=False in dronekit.connect function the problem disappeared. I don't know this can help.
Thank you,

Gabriele

gab933, 2021-05-12:
Hi,
I use exactly the same to connect. Sorry for my redundancy, I'm able to explain more in details the problem:
Every operation is performed by keeping Rosetta **active**:
- 1st dronekit connection always works
- 2nd dronekit connection sometimes works
- 3rd to 4-5th connection fails
- 5th to n-th connection fails, and Rosetta heartbeat becomes discontinuous (dronekit logs "No heartbeat in 5s/ link restored").

By stopping and re-launching Rosetta I'm able to connect successfully (we fall inside the 1st connection case).
Instead, when using SITL, same procedure, everything works (just to have a comparison).
I attempted to set wait_ready=False inside connect() and remove wait_ready('autopilot_version'), and connection always succeeds! Dronekit telemetry is always available by using this settings, the problem resides in dronekit commands: they sometimes work, sometimes are delayed, sometimes are not received at all!

gab933, 2021-05-12:
Hi,
I've further investigated on connection problem:
by setting wait_ready=True is the same as setting wait_ready=['parameters', 'gps_0', 'armed', 'mode', 'attitude']).
By making some test, I can confirm that by omitting 'parameters' field connection immediately establishes. Therefore, more the connections we establish (keeping Rosetta **up**) more the time Rosetta uses to send parameters, ending on connection timeout!
Even if we omit 'parameters' field and therefore by establish a rapid connection, commands sent to Rosetta are not received until parameters are correctly sent. Since the time for sending those parameters increases with the number of connections, commands sent to Rosetta are progressively delayed (I test a set_servo command has been processed after about 5 mins I sent it from dronekit).
Hope this can help,
Regards,

Gabriele

The1only, 2021-05-12:
Ahhh. Ok... interesting, sending the parameters only take a sekund or two in my system.
I never checked if is sends it to old sockets.

Best regard
Terje Nilsen
9Tek AS
Norway

gab933, 2021-05-13:
I forgot to specify that I'm using work branch "version" of Rosetta, I don't know if there is some misalignment with respect to master branch.
Regards,

Gabriele

gab933, 2021-06-03:
Hi, solved.
By reinstalling the app everything works as expected. Maybe the installation via Android Studio generated this issue, now I've exported the apk file on sd card.
Thank you,

Gabriele

closed 2021-06-03 by gab933