RosettaDrone/rosettadrone #71: Error in Android APK

cepoya, 2021-05-17:
After compile in android studio and create apk file.
the drone joystick already connected with mobile phone (i used huawei mate pro 20) and vehicle already connected with joystick.
unfortunately, the mobile application (Rosetta drone 2) not working well (doesn't detect the joystick and vehicle)
[image]

could anyone help my problem?
thanks

The1only, 2021-05-17:
Did you shut down Go-4, you can only have one app controlling the drone.

Best regard
Terje Nilsen
9Tek AS
Norway

cepoya, 2021-05-18:
> Did you shut down Go-4, you can only have one app controlling the drone. Best regard Terje Nilsen 9Tek AS Norway
> […](#)

al

cepoya, 2021-05-18:
> Did you shut down Go-4, you can only have one app controlling the drone. Best regard Terje Nilsen 9Tek AS Norway
> […](#)

DJI GO 4 already shutdown but Rosetta Drone 2 APK, still doesn't work

cepoya, 2021-05-18:
<meta-data
            android:name="com.google.android.geo.API_KEY"
            android:value="@string/google_key" />

how i get google key?

cepoya, 2021-05-18:
update--

already connected with drone (but the Model not available)
and the APK don't wanna show and restart the application in android

[image]

cepoya, 2021-05-20:
https://user-images.githubusercontent.com/62041698/118914033-baf5e280-b954-11eb-85b7-20a7080ad192.mp4

still error, anyone can help me?

cepoya, 2021-05-28:
after disable rotate display,
the application already run

closed 2021-05-28 by cepoya

davidan208, 2023-08-03:
hi @cepoya,
Can you help me with the way to build on the Android studio.
I had initialized the git lfs, created new key.xml file with dji key and google map key, but when I tried to build, I got this problem.
[image]
How did you overcome this problem, I downloaded the lastest version of Android Studio but it didn't work.
Thanks.