RosettaDrone/rosettadrone #73: Video not shown [work branch]

VladimirPobedinskiy, 2021-05-18:
Video not shown in videoframe. Only label "WAITING FOR VIDEO".  _[work branch]_
I try all options nad ports. It not help.
I don't know where problem - in Rosetta or in QGroundControl.

The1only, 2021-05-18:
> I try all options and ports. It not help.

Use netcat to dump the UDP stream to verify if Rosettadrone sends data.

Best regard
Terje Nilsen
9Tek AS
Norway

cepoya, 2021-05-31:
> Video not shown in videoframe. Only label "WAITING FOR VIDEO". _[work branch]_
> I try all options and ports. It not help.
> I don't know where problem - in Rosetta or in QGroundControl.

same with me, the video not show in qground control in my mobile and in my laptop

Phillweston, 2022-01-13:
I have also encountered this situation several times, and the result of the later test is that after opening the QGC ground station and then opening the rosettadrone APP, you can see the video screen at the ground station.

kripper, 2023-04-08:
See https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/edit#heading=h.q9mp6yjsfxe5

closed 2023-04-08 by kripper