RosettaDrone/rosettadrone #74: Altitude Problem with QGC in Simulation

cbayerlein, 2021-05-19:
**Describe the bug**
There is a problem with the altitude display and automatic altitude hold in conjunction with QGC in simulation mode on Rosetta Drone 2.

I'm trying to run Rosetta Drone 2 in simulation mode to see if my set-up works and if it's safe to fail.

If I start RD 2 alone, everything works fine. I can take off and move with the joysticks on the remote control.

But as soon as I start QGC, there are problems with the altitude. The simulated drone climbs higher and higher without end. The height display in QGC also has a decimal point (it always shows 100 times the height that RD 2 shows).

As long as I'm on the ground everything is fine. The problem always starts after the start, regardless of whether I start from RD 2 or whether I use the start function in QGC for this.

So far I have only tested the functionality in simulation mode. After discovering this error, I don't dare to open QGC in real operation and connect it to RD 2. I use QGC as an external gcs on windows.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the RC to the smartphone and power on RC and drone.
2. Start RD2 on the smartphone
3. Select simulation by clicking "simulate flight"
4. Click on Open
5. Click on the gear icon and then on settings to set up external GCS
6. leave settings menu
7. disable safe mode by clicking on the padlock icon

from here there two ways to reproduce:

A)
8. Take off by clicking the icon in RD2 and confirming with the slider
9. The drone takes off and hovers at 1.2m over ground
10. Start QGC on the external machine
11. RD2 connects to QGC successfully (connection announced and map is updated in QGC and GCS goes green in RD2)
12. The drone starts to climb higher and higher. Altimeter in QGC shows wrong altitude (10x the alt in RD2)
13. Quit QGC. 
14. The drone hovers again

B)
8. Start QGC on the external machine
9. RD2 connects to QGC successfully (connection announced and map is updated in QGC and GCS goes green in RD2)
10. Take off within QGC by clicking the icon there and confirming with the slider
11. The drone takes off and immediately starts to climb higher and higher. Altimeter in QGC shows wrong altitude (10x the alt in RD2)
12. Quit QGC. 
13. The drone hovers 

**Expected behavior**
A12 / B11 should not happen. Instead, the drone should hover over ground after take-off. Altimeter in QGC should be correct

**Desktop (please complete the following information):**
 - Win10
 - QGC build 2021-05-06

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S8
 - OS: Android9
 - RD2 version 5.1.1

**Additional context**
As I said, I tried this setup at home in simulation mode. Both, smartphone and PC are connected via WIFI. I tried RD2 in a real world scenario (not simulated) without QGC before and everything worked fine then. I also tried QGC with PX4 SITL and that also worked fine without the described altitute problems.

cbayerlein, 2021-05-19:
I just discovered that when I disable both the virtual joystick and the hardware joystick in QGC, the ever increasing altitude issue doesn't appear. Sorry, forgot to write down that I activated this.

However, the problem with incorrect altitute display in QGC still persists.

Also, having to disable the joysticks is suboptimal. That was actually my use case: due to my physical handicap, I use a special joystick that identifies itself as a game controller on the PC. I would also like to be able to control the drone with it.,

The1only, 2021-05-19:
Yes this is know, but it is so easy to fix most of us just fix it and noone has ever pushed the fix.
Find elevation report and do the math.

I will fix this together with a loot of other fixes this summer, when things comes down. 
I am flying long (4.5KM +)  missions with DJI Mavic 2 Pro and QGC these days, works fine. But still a few bugs inn loading a mission several times etc.

Best regard
Terje Nilsen
9Tek AS
Norway

cbayerlein, 2021-05-26:
Thank you, Terje! I found the error in the code and fixed it with this commit:

https://github.com/cbayerlein/rosettadrone_mini2/commit/16f4c253bc5336e77d9e5e8d31a0cf143fdaa714

However, since the altitude is an integer, I can't see cm values in QGC. Is there a way to fix this?

The1only, 2021-05-26:
Yes,

/**
 * This is the altitude above the home position. It resets on each change of the current home position.
 */
public float altitude_relative;

private void send_altitude() {
    msg_altitude msg = new msg_altitude();
    LocationCoordinate3D coord = djiAircraft.getFlightController().getState().getAircraftLocation();
    msg.altitude_relative = (float) (coord.getAltitude() ); //* 1000);
    m_alt = msg.altitude_relative;
    sendMessage(msg);
}

Inn the Mavlink message it is already a float…

cbayerlein, 2021-05-26:
Great, thanks for the hint! Changed int to float in this commit:

https://github.com/cbayerlein/rosettadrone_mini2/commit/4526f9ee61c54f169e6d8c04afaebe013f3cc53c

cepoya, 2021-05-28:
> **I edited this issue because originally, I reported two problems in one issue. Now I seperated them to make them clearer**
> 
> **Describe the bug**
> There is a problem with the altitude display in conjunction with QGC on Rosetta Drone 2.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> 1. Power on RC and drone.
> 2. Start RD2 on the smartphone click on Open
> 3. disable safe mode by clicking on the padlock icon and take off by clicking the icon in RD2 and confirming with the slider
> 4. Start QGC - RD2 connects to QGC successfully (connection announced and map is updated in QGC and GCS goes green in RD2)
> 5. Altimeter in QGC shows wrong altitude
> 
> **Expected behavior**
> Altimeter in QGC should be correct

can you give link the QGC master?
the drone can't communicate (can't connect and update firmware with QGC)

closed 2023-02-17 by kripper