RosettaDrone/rosettadrone #75: As this project grows we need a small team to manage the code.

The1only, 2021-05-19:
As this project grows more and more popular, I can not keep up all by my self. We need a small team of 3-4people willing to discuss strategy and directions, to accept PR's accordingly and to answer questions.

Do we got any volunteer??? 
I’ll keep doing my best....

m4xw, 2021-05-27:
Sure, sign me up.
Any preferred Platform for Discussion (Discord?)
I am very limited by time tho in regards to making active code changes so I'd like to keep them in scope of whatever I am working on at the moment.

The1only, 2021-05-27:
Then we are in the same position.

I’ll wait untill we are 4 people and then I call for a Teams chat…

Terje

m4xw, 2021-05-27:
Yea sounds good.
Its not like the project is running away :P
I def have some ideas where we could go from here. I'd like to split the DroneModel up and also do a util class for the GPS math helpers.
Also the waypoint stuff should be abstracted a bit differently and we need some better system to handle tasks instead of random timers. Surprisingly enough most stuff of the SDK seems fine when being called from another thread, so maybe dedicated worker threads are the solution here & some basic event system where u can register/unregister from.
I think most of us just hardcode stuff in rosetta anyway so it would make sense to design it to use some form of library internally instead of having dead funcs all over.
That would probably cause a lot of conflicts with people and their local forks when they try to rebase tho.

cbayerlein, 2021-05-27:
Yes, I'm happy to help too, although I'm not sure how much I can actually contribute. I am not an expert (neither in relation to DJI SDK, nor Android development or MAVLink), but I rather use try and error approach with a little background knowledge of coding (I studied computer science years ago, therefore reasonably good overview knowledge, sometimes it is lacking on the specific details).

Regarding my implementations in the area of accessibility: yes, although I will first concentrate on my own use case (I would like to be able to operate the drone with the assistive technologies that are available to me and that I can use, especially my micro joystick).

Accessibility in itself is a broad field and you could do a lot there, but that also means a lot of work.

Polymathical, 2021-06-18:
You write really clean code. I'd be honored to be a part of the project

Phillweston, 2022-01-13:
I've been working on this project recently, and we're working on some industry-grade features on top of the DJI Mavic 2 Enterprise Advanced model, as well as some open-source flight control code for Ardupilot.
I'd be honored to be a part of the project.

BkMak16, 2022-01-24:
I'm here, time limited like the rest of us but absolutely here.
I managed to clear up the things on my plate and would be awesome to part of this.

Python all the way.

kripper, 2023-01-09:
Same here. I'm interested in writing AI scripts for the Mini SE.
I would like to see you all merging your branches so we can continue working on a single code base.

kripper, 2023-02-17:
Duplicate. Closing.
Please everybody follow our conversations [here](https://github.com/The1only/rosettadrone/issues/19).

closed 2023-02-17 by kripper