RosettaDrone/rosettadrone #77: the unit of bitrate in VideoService

lukelu0520, 2021-05-25:
Dear rosettadrone, 

In VideoService.java, there is a parameter mvideoBitrate set to 3000
Is the unit bps or kbps ?

Hope to hear your expertise.

The1only, 2021-05-25:
The value 3000 is never used, it is set from the config. Default is 2mbit. However this only applays to some drone types, end even there it seems not to do much.

https://developer.dji.com/api-reference/android-api/BaseClasses/DJIVideoFeeder.html#djivideofeeder_settranscodingdatarate_inline 

Best regard
Terje Nilsen
9Tek AS
Norway

lukelu0520, 2021-05-25:
Thanks for your reply @The1only , then what's the bitrate used by rosettadrone for streaming to QGroundcontrol ?

The1only, 2021-05-25:
Depends on drone.
Let us know when you find out.

Best regard
Terje Nilsen
9Tek AS
Norway

lukelu0520, 2021-05-25:
We have i) Mavic mini and ii) Mavic Air 2

himanshu131098, 2022-02-11:
Does the Video transfer not work if the `mVideoBitrate` and `mEncodeSpeed` are not configured according to the drone. If yes, are there any references to get these values for the particular drone?
@The1only @lukelu0520 

Thank You

Himanshu V
Vyorius Drone Pvt Ltd
Delhi, India

kripper, 2023-04-08:
Yes, bit rate is completely ignored.

closed 2023-04-08 by kripper