RosettaDrone/rosettadrone #78: Manual control with virtual joysticks in QGC leads to ever increasing altitude

cbayerlein, 2021-05-26:
If I activate the virtual joysticks in QGC and activate the checkmark for the automatic centering for throttle, then the drone continues to climb in height.

I debugged the behavior in the code a bit and found that QGC sends "manual control" messages when the virtual joysticks are activated. A thrust of 500 is transmitted if the setting is active that the throttle is centered. This is transmitted to the controller in RD 2 as a positive value for the Z-axis (approx. 1.92 ..) and thereby the drone continues to climb.

If I deactivate the automatic centering, a value of 0 is correctly transmitted for the thrust (no change in height). However, I can no longer fly downwards with the thumbpad.

I don't know whether the problem is with RD2 or with QGC. On the one hand, the MAVLink protocol specifies a value range from -1000 to +1000 as valid. So the middle should be 0. Thus, QGC would be out of specification. On the other hand, I have already tested QGC with other firmwares (PX4) with the virtual joysticks and everything worked here. It could of course be that PX4 implements the MAVLink specification incorrectly as well, but I think that is unlikely.

What would be the best way to get the behavior right?

The1only, 2021-05-26:
Interesting, I supose it must be a mode thing that is missing. I will check.

Best regard
Terje Nilsen
9Tek AS
Norway

cbayerlein, 2021-05-26:
Cool, thank you so much!

cbayerlein, 2021-05-27:
I just had a quick in QGC code. Obviously, it sends only positive values (between 0 and 1000) for throttle, except for rovers:

https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightDisplay/VirtualJoystick.qml#L44

I'm not sure if this behavior conforms to the MAVLink specification for quadcopters, but at least this approach seems to be compatible with PX4. In the SITL I can't see that the virtual drone would keep increasing in altitude.

I still have to find out what happens with hardware joysticks, especially again what the value range looks like there. I'll try that out tomorrow.

The1only, 2021-05-28:
It’s a mode thing. If the throtle stick controls motor throtle then ther is no negative values, one can never have negative throtle on multirotors. However if it controls altitude rate, then it can be centered and the value can be positive and negative and represent rate of change of altitude. So the only way to fly with the throtle stick centered it to make sure one use correct mode, and there must be something there….
It this is a flight mode thing, or a bit in the mavlink command, I have yet to work out.

Best regard
Terje Nilsen
9Tek AS
Norway

cbayerlein, 2021-05-28:
Ok, for now I changed MAVLinkReciever to correct z value:

https://github.com/cbayerlein/rosettadrone_mini2/commit/4bfd6e2d37efcf921627f720b2ca53c6c2f4b40e