RosettaDrone/rosettadrone #79: Wrong Yaw behaviour when using manual control

cbayerlein, 2021-05-28:
Due to a bug in DroneModel.do_set_manual_velocity the yaw control did not work properly. If you use (virtual) joysticks in QGC, the drone has automatically pointed to the north again and again. The cause was a wrong yaw control mode. I corrected this here:

https://github.com/cbayerlein/rosettadrone_mini2/commit/f171f607e6ee0e9c316c8ce2dd5dd3676d4157d0

m4xw, 2021-05-31:
Thats a issue in my fork on not a upstream issue.
It was intended to work with absolute angle values and not velocities in my fork

m4xw, 2021-05-31:
I did some corrections here https://github.com/m4xw/rosettadrone_mini2/commit/59fa4d74ea9a796d1d9506d98816b1cfdef0cee1

cbayerlein, 2021-05-31:
Oh, I see. sorry then and thanks for the corrections, @m4xw !

closed 2021-05-31 by cbayerlein

m4xw, 2021-05-31:
I also did fixes for the actual GoTo via the GCS, so make sure to rebase.
if gcs sends absolute angles we might need to apply this here:
[image]

m4xw, 2021-05-31:
(So you can basically move via right click now)