RosettaDrone/rosettadrone #82: Maps shows no google services on smart controller

highfreq, 2021-06-28:
I have everything compiled and working, my only prob at the moment is the maps rectangle on the smart controller screen showing "Remote stream won't run without Google play services wich are not supported by your device".

Any solution to that? at least not showing rectangle.

thanks

The1only, 2021-06-28:
You need a google maps account and a key for it to work.

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
Thank you, i have a key that includes maps, how can i put it in the project to compile with support for maps?

highfreq, 2021-06-28:
Never mind found it!!!!

Should have look a bit better :)

The1only, 2021-06-28:
The key goes in the same file as the DJI key.
Do not have the name right now.

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
In android studio should i see the smart controller as rm500  ??

The1only, 2021-06-28:
YES

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
Don't know why but i am not able to see it anymore. I have debug enabled but it doesn't show in android studio. I tried to revoke usb permissions but it doesn't ask for them anymore on usb connection. Phone connects and can see in Studio, just problem with smart controller. Any hint?

The1only, 2021-06-28:
Change the mode in the controller to «charge only» if that does not help change is back to «file mode»… this must be done while connected to the USB bus. Some times one must reboot everything.

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
rebooting everything seems to have solved the prob, thanks

highfreq, 2021-06-28:
I put my key in and now i get message "Rosetta Drone 2 won't run wothout google play services wich are not supported by your device". I get this buy running the app from android studio with device attached. 
If instead of smart controller i use my samsung s21 phone than i get maps showing.

The1only, 2021-06-28:
Chinese device? 
Must have google.

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
It is a dji smart controller

highfreq, 2021-06-28:
Solved it installing google services, now i have playstore too :)

Only problem left i get no video neither on the smart controller nor on the pc ground station, telemetry connects.

The1only, 2021-06-28:
Yes Chinese, no Google… you must add Google manually. Get the .apk online.

Best regard
Terje Nilsen
9Tek AS
Norway

The1only, 2021-06-28:
What drone are you using? And what compiler tool version? 

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
Mavic pro 2. I installed Google services and all good on maps now.

Please let me know where to look for all info on android studio. Actually i think an update i made to android studio broke rosetta video.

highfreq, 2021-06-28:
Android Studio but maybe is gradle but i don't know where to look

highfreq, 2021-06-28:
In File->project structure->project:

Gradle plugin 4.2.1
Gradle version 6.7.1

The1only, 2021-06-28:
Yes you can NOT update, no one have had time to figgure out why.!

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
So what can i do to fix it?

The1only, 2021-06-28:
You need 6.3.x

We do not know why, needs to be debugged.

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
I selected gradle 6.3 in project and tried to build but it says minimum version supported 6.7.1
Would you be so nice to tell me if and what i need to select to compile?
thank you very much for your patience

The1only, 2021-06-28:
Was it the pluggin that needs to be 3.6? 
Check out in a new catalog and verify the origional files.

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-06-28:
I don't know how to do it. Is there a simple way of telling me what plugin and gradle to use?

highfreq, 2021-06-28:
i downloaded project again and compiled, now it works.

Plugin 3.6.4
gradle version 6.7.1

closed 2023-02-17 by kripper