RosettaDrone/rosettadrone #87: Joystick behavior

highfreq, 2021-07-13:
Hello, is it normal that once i take off with the regular transmitter if i have the joystick attached to ground station, joystick takes control and i can't do anything anymore with transmitter?

I have to say that i am testing rosetta 2 with a spark that maybe is not supported and what i m seeing is just a matter of a not supported drone.

The1only, 2021-07-13:
Normally the transmitter shuld override any other controll. As soon as you tutch the controller stick the system should make that in command. However I have also seens some issues around this in AI mode.

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-07-14:
Whatdo you mean by AI mode?
I tested it only inside without gps and it does it every time.
I also have aanother problem, the throttle doesn't work from the joystick, once is in the air the only wy is to lower transmitter control, joystick only goes up, not down. I tried calibrating it in every possible way, with throttle on low stick and throttle on center stick but it never works on joystick

The1only, 2021-07-14:
I belive you now are at the same point as the rest of us. The throtle down is a challenge.

Best regard
Terje Nilsen
9Tek AS
Norway

highfreq, 2021-07-14:
Ah ok :)

kripper, 2023-02-18:
When you touch the RC, the drone changes to STABILIZE flight mode and is supposed to override other commands (including VirtualStick).
When you send new commands via MavLink, it should go back to GUIDED flight mode.
In Rosetta 3, we call `enableVirtualStickMode()` all the time to make sure VirtualSticks are re-enabled.

I played with VirtualSticks and had no problem in going up or down, but the drone (DJI MINI SE) has a sensor.
Under 0.5 [m] you need to disable the landing protection with `setLandingProtection(false)` to be able to descend furher.

Closing for now.

closed 2023-02-18 by kripper

Sasysan, 2023-03-27:
I'am sorry to open this issue again but, i notice that with my drone ( DJI Mini 2 ) when i am in STABILIZE i can't use the RC. I mean i try to change the yaw or go up or down but the drone doesn't respond any solution?
Thanks

kripper, 2023-03-27:
When visual sticks are enabled, the RC actions are ignored 

Add code to disable the virtual sticks on the method that cancels all motion (usually triggered when the destination point has been reached).

Place a breakpoint on the code that enables virtual sticks to check it is not re-enabling them.