RosettaDrone/rosettadrone #91: Where to find the list of MAVLINK msgs supported and publishing frequency

frontw, 2021-08-23:
Hi! 

I just found this interesting project and trying to understand the possibility of its usage for drone control to fly along a given trajectory with ROS from PC.

Where could I get the list of data I could obtain? Could I get the position, orientation, linear and angular velocity?
What is the frequency of this data? Could it be 50 Hz or something like this?
How I could control the drone?

Thanks a lot!

frontw, 2021-08-23:
Looks like I could find whar rosettadrone receives and sends in app/src/main/java/sq/rogue/rosettadrone/MAVLinkReceiver.java and app/src/main/java/sq/rogue/rosettadrone/DroneModel.java
Am I right that maximum frequency is 10Hz? (according to `void tick() { // Called ever 100ms...`)?

The1only, 2021-08-23:
Yes,  can be changed if you need faster updates, but it is a CPU tradeoff.

frontw, 2021-08-23:
Thanks for answering!
According to your practice, does Mobile SDK itself let to get new data at a frequency around 50 Hz?

The1only, 2021-08-23:
I did not try but it can be set.
However I fly using computer vision AI in a PC on the ground and Dronekit as a controller all the time, and 10Hz is no problem, remember the drone is actually flying, all you need to to correct it’s heading/altitude/direction etc.
Unless you whant to fly very fast, then 50Hz make sense.

kripper, 2023-02-17:
Check the code. Search for "MAV_"

closed 2023-02-17 by kripper