RosettaDrone/rosettadrone #92: Can't build due to linking failure: FAILED: libdjivideojni.so

laurenweinstein1, 2021-09-28:
Builds (on all platforms) are failing at linkage with:

FAILED: libdjivideojni.so

I've seen this discussed in some other rosettadrone forks but the solutions seem nebulous at best and not clear what the best solution is for this fork. Any ideas? Does anyone have a working apk (with the current Mini 2 support) available for testing? Thanks very much!

himanshu131098, 2022-01-27:
Make sure you use `git lfs` installed in your system and clone it using the commands:
```
git lfs install 
GIT_SSL_NO_VERIFY=1 https://github.com/The1only/rosettadrone.git
```

himanshu131098, 2022-01-27:
Yes, I think you should add it in the README.md. 

Thanks & Regards
Himanshu V
Vyorius Drones Pvt Ltd
India

closed 2023-02-17 by kripper