RosettaDrone/rosettadrone #96: Help to connect drone to python program

kumar-rajwnai, 2021-11-24:
I have seen dji tello but i want to use DJI Mavic series.

https://diydrones.com/profiles/blogs/dji-drone-controlled-by-ai-python-script

I want to create this thing. I have no idea about how to connect drone to my python application. Can you give me an overview of how to get frames from the drone in my python application and control drone?

Is there any article which teach this things from scratch then please also share the link of it.

BkMak16, 2022-01-24:
Is it possible, yes.

Is there an article or tutorial, no.

What you're referring to is my contribution to this and we're still working on it. It might take some time.

Welcome to the Think Tank.

The1only, 2022-01-24:
There are examples in this Q&A in some stream.

Best regard
Terje Nilsen
9Tek AS
Norway

kripper, 2023-04-08:
See https://github.com/RosettaDrone/rosettadrone/discussions/133

closed 2023-04-08 by kripper