RosettaDrone/rosettadrone #97: Add support for Autel EVO II series (also Nano and Lite)

GlidLov, 2021-12-10:
Pls add Autel EVO II series support, they already have an SDK with commands and video stream avaiable (ask manifacturer)
They are better than DJI , IMHO

The1only, 2021-12-10:
Hmm what makes it better?
Looks like a low-cost copy. Is it US made?

Best regard
Terje Nilsen
9Tek AS
Norway

GlidLov, 2021-12-13:
no no-fly zones
6 axis avoidance
great SDK
*much* better cam (and more options of that)
no spying chinese gov backdoor (at least until now)
enterprise edition avaiable (more accessories, RTK)
US company, China made
better price

The1only, 2021-12-13:
The Rosettadrone is very tight copled to the DJI SDK, supporting a new arcitecture will be some job. Are the EVO company willing to support a drone to the comunity if we do this?

Best regard
Terje Nilsen
9Tek AS
Norway

GlidLov, 2021-12-13:
I have no idea if Autel will give one drone to the community, i'm just a user which have requested the SDK and had a look at test app (which shows main sdk functions). I do not know the DJI sdk but the autel one seemed to me very linear and simple, with good docs and examples.

patogit, 2022-02-21:
Donating one drone seems like a pretty easy ask if the result is getting QGroundControl support via RosettaDrone. However, Autel has stated that their Nano and Lite drones will run with an app that doesn't do waypoint missions. Their larger drone, the Evo II, uses the Autel Explorer app and can plan and fly missions from within that app. Their market strategy seems to be: if you want to map, buy the big drones, and if you want to fly something small and make nice videos, buy the small drones. Given that they've taken this approach, they might not care that people want to fly missions with their small drones. I don't really understand their logic, but then again, so many manufacturers of so many things publish software that doesn't utilize the real capabilities of the devices (drones, cameras, routers, cell phones, etc.).

patogit, 2022-02-21:
In terms of "what's the difference", Autel is pretty good quality, at least on par with DJI if not better in some ways. There are multiple videos of people pouring water on the Evo II or crashing it into a river, then drying it out and flying again with either no damage or only battery damage. And there's the Dragonfish line, but no one here has enough money for that, unless you find a few bitcoins under your sofa cushion.

kripper, 2023-02-17:
Would be fun to do and a good change to centralize all model related code into DroneModel.java...but we have no model for testing at the moment.

closed 2023-02-17 by kripper