RosettaDrone/rosettadrone #99: Add support for Mini 2

lmore377, 2022-01-01:
DJI finally released v4.16 of the Mobile SDK with support for the Mini 2. I tried it out by editing /app/build.gradle and changing the following dependencies (these are the only changes I've made):

```
com.dji:dji-sdk:4.14-trial1 -> com.dji:dji-sdk:4.16
com.dji:dji-sdk-provided:4.14-trial1 -> com.dji:dji-sdk-provided:4.16
com.dji:dji-uxsdk:4.14-trial1 -> com.dji:dji-uxsdk:4.15
```
Working (Partially or completely):
- Flight Simulation
- Telemetry from drone (Very incorrect values are being sent to QGroundControl but everything looks fine in the app)
- Joystick events from QGroundControl (I'm not able to decrease altitude but I think this might be a GCS problem. Only tested with simulator)

As for what's not working/problematic, it has pretty much all the same problems as this fork: https://github.com/m4xw/rosettadrone_mini2

I'm planning on adding some of the patches from that repo to this one to see if things start working better.

lmore377, 2022-01-01:
So I tried changing the SDK version on that fork and it fixed the telemetry issues but video is still broken. logcat gets spammed with `H264Packetizer: NAL units are not preceeded by 0x00000001`

laurenweinstein1, 2022-01-01:
Be sure to use the "change_outputsurface" branch from https://github.com/m4xw/rosettadrone_mini2

This does work for video, but can require toggling between photo and video mode in the app to get it going in various modes.

lmore377, 2022-01-02:
Yep that worked. Thanks for the tip!

m4xw, 2022-01-03:
Bummer they didnt fix that shit.
Either way, will check out the SDK in a few days and integrate the changes from my branch + general cleanup.
Feel free to take it as a base.

m4xw, 2022-01-03:
Extra notes: Fwiw DJI Fly app uses v5 SDK and does not exhibit the issue with DJI's decoder, so i did some shenans that use the dji video decoder via mediacodec and then passes it to ffmpeg so u can also stream to GCS at the same time and it also bypasses the issue. However that still exhibits the same issue like the DJI Fly app on my S21 (exynos), essentially breaking 4k30 streaming with hwcodec, the pure ffmpeg variant does not exhibit the behaviour but also doesnt produce new keyframes (alternatively can implement that in C but then always need to keep a full chain of frames around which is expensive for low end devices, memory wise)

m4xw, 2022-01-03:
>Telemetry from drone (Very incorrect values are being sent to QGroundControl but everything looks fine in the app)

For the other mentioned issues (fixed on my fork), see https://github.com/The1only/rosettadrone/issues/74
https://github.com/The1only/rosettadrone/issues/78

lunayap, 2022-03-18:
@m4xw have you had a chance to look into mini 2 video feed stream to GCS?

lunayap, 2022-03-18:
In previous comment you said:
> so i did some shenans that use the dji video decoder via mediacodec and then passes it to ffmpeg so u can also stream to GCS at the same time and it also bypasses the issue

Do you have a working example for this? I really appreciate your help cos I have zero experience with codecs and definitely couldn't wrap my head around the one KeyFrame inserted in the beginning

m4xw, 2022-03-21:
> In previous comment you said:
>
>
> Do you have a working example for this? I really appreciate your help cos I have zero experience with codecs and definitely couldn't wrap my head around the one [...]

Yes, its all on my "change_outputsurface" branch https://github.com/m4xw/rosettadrone_mini2/tree/change_outputsurface
Weather has been horrible here and my flight location is a goner rn, so dev has currently stagnated there.

The1only, 2022-05-23:
DJI SDK 4.16.1 now default.

closed 2022-05-23 by The1only

m4xw, 2022-05-23:
👍 Once I feel like working on some drone stuff again (no location to fly rn), I will update my PR with all other changes/fixes like the Waypoint interpreter and the fixed video decoding + dual video output for app and gc
So please leave that open for now ;)

patogit, 2022-06-01:
I just built the master branch from this repo, and ran it with my Mini 2, and it shows a live video feed in QGC on the same phone, even though "video feed" is disabled. I can take photos. I can disable safety. I can see telemetry. But I haven't yet figured out how to fly a survey mission planned in QGC, or control the drone's movement in any way from QGC.