RosettaDrone/rosettadrone #101: Build Failure on Linux, with ARM architecture

NuclearManD, 2022-01-23:
**Describe the bug**
Build failing for ARM

**To Reproduce**
Connect Samsung Galaxy A52 to computer, enable debugging
Attempt to run the app on the phone

**Expected behavior**
App builds, launches, and runs normally

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux Mint
 - Browser Brave and Chrome

5.11.0-34-generic #36~20.04.1-Ubuntu SMP Fri Aug 27 08:06:32 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy A52
 - OS: Android
 - Browser: Samsung Internet
 
**Additional context**

Error messages, as text:

```
Error while executing process /home/nuclaer/Android/Sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /home/nuclaer/StudioProjects/rosettadrone/app/.cxx/cmake/debug/arm64-v8a djivideojni}
ninja: Entering directory `/home/nuclaer/StudioProjects/rosettadrone/app/.cxx/cmake/debug/arm64-v8a'
[1/1] Linking C shared library ../../../../build/intermediates/cmake/debug/obj/arm64-v8a/libdjivideojni.so
FAILED: ../../../../build/intermediates/cmake/debug/obj/arm64-v8a/libdjivideojni.so 
: && /home/nuclaer/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=aarch64-none-linux-android22 --gcc-toolchain=/home/nuclaer/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/nuclaer/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libdjivideojni.so -o ../../../../build/intermediates/cmake/debug/obj/arm64-v8a/libdjivideojni.so CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o  ../../../../src/main/cpp/lib/arm64-v8a/libffmpeg.so /home/nuclaer/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/aarch64-linux-android/22/liblog.so -latomic -lm && :
ld: error: ../../../../src/main/cpp/lib/arm64-v8a/libffmpeg.so:1: unknown directive: version
>>> version https://git-lfs.github.com/spec/v1
>>> ^
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

NuclearManD, 2022-01-24:
I also have this warning:

```
ABIs [arm64-v8a,armeabi-v7a,armeabi] set by 'android.injected.build.abi' gradle flag contained 'ARMEABI' not targeted by this project.
```

The1only, 2022-01-24:
You did not have Git LFS installed when you checked out the files from git.
I stall Git LFS and start all over.

Best regard
Terje Nilsen
9Tek AS
Norway

NuclearManD, 2022-01-26:
Thank you!  Ah, you are right, I did not have it installed.  I installed it, re-cloned the repo, and attempted app install again, and it worked!

Thanks!

closed 2022-01-26 by NuclearManD