RosettaDrone/rosettadrone #102: Waypoint missions with QGroundControl

gsenkowski, 2022-02-03:
After uploading a waypoint mission with QGroundControl, the message 'Mission transfer failed. Error: Mission write failed, vehicle failed to send final ack.' is shown in QGroundControl.

The log of rosettadrone shows 'Mission uploaded and ready to execute!' and after that 'Mission status: 0'

Steps to reproduce the behavior:
1. Start UAV in a place with GPS reception, connect Remote to Smartphone
2. Launch Rosettadrone
3. Launch QGroundControl, on PC or smartphone
4. Create a valid (within the limitations of the dji mobile sdk) waypoint mission
5. Upload the mission to the UAV
6. see 'Mission transfer failed. Error: Mission write failed, vehicle failed to send final ack.'

**Expected behavior**
The mission is uploaded to the UAV and can be launched from QGroundControl

PC:
 - OS:Ubuntu 18.04, Ubuntu 20.04
 - QGroundControl 4.0.11, 4.2.0, 

Smartphone:
 - Device: Nokia 7.2
 - OS: Android 10
 - Rosettadrone current master branch
 - QGroundControl v4.1.1 64 bit

The1only, 2022-02-03:
This was fixed in a not yet released version.
It had to do with mission sequensing.

Best regard
Terje Nilsen
9Tek AS
Norway

Phillweston, 2022-02-04:
I'm having the same problem.
I try to do as follow:
1. Start UAV in a place with GPS reception, connect Remote to Smartphone
2. Launch QGroundControl, on PC or smartphone
3. Launch Rosettadrone
4. Create a valid (within the limitations of the dji mobile sdk) waypoint mission
5. Upload the mission to the UAV
Sometimes the problem is not reported in error

MariusBeul, 2022-02-08:
I am having the same problem. @The1only can you point what is wrong in the current version?

Phillweston, 2022-05-23:
> I am having the same problem. @The1only can you point what is wrong in the current version?

Me too, I found that reinstalling app can solve the problem, but soon the problem comes.

The1only, 2022-05-23:
I will see if I can get out a fix later today. It will however also contain a series of other patches also. Than you for the reminder.

Best regard
Terje Nilsen
9Tek AS
Norway

closed 2023-02-17 by kripper