RosettaDrone/rosettadrone #103: Unable to get Video feed to Ground Station while using MAVIC_2_PRO & PHANTOM_4_PRO_V2

himanshu131098, 2022-02-04:
I have tried to build APK with both `master`  and `work` branches but was unable to get the video stream on the Ground Station for both **MAVIC_2_PRO** and **PHANTOM_4_PRO_V2**.
Any help with this will be much appreciated.

Thank You

Himanshu V
Vyorius Drones Pvt Ltd
Delhi, India

Phillweston, 2022-03-26:
I have encountered a similar problem before. If the APK is in the background after running, the UDP video stream of QGC will be interrupted, or other third-party applications may also cause the video stream to be interrupted when running.

kripper, 2023-02-17:
Please check if the video streeam **on QGC** works for you on [this fork](https://github.com/RocketRider/rosettadrone)

kripper, 2023-04-08:
Also see:
https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/edit#heading=h.q9mp6yjsfxe5

closed 2023-04-08 by kripper