RosettaDrone/rosettadrone #105: Internet disconnection

gab933, 2022-02-23:
Hi,
I made the following test:
* Start application with Internet connection on (everything works as expected)
* Disconnect Smart RC from Wi-fi (emulating Internet disconnection)
* Connect Smart RC to Wi-fi again (all ops have been performed with app still running)

I noticed telemetry is not anymore received on my GCS.
Instead, commands from GCS are correctly sent to Rosetta, but drone does not execute them.
I attach a log example.
At 10:06:00 I disconnected Wi-fi, and an error appears:
"set_timerslack_ns write failed"
.
.
At 10:06:32 I connect again Wi-fi and same error as before appers, together with:
"org.json.jsonexception"
.
.

How can I fix this issue, in such a way I'm able to manage disconnection?
Thank you

[image]

The1only, 2022-02-23:
I have seen this too. Maybe it’s time to fix it.
I belive it is something about the order of UDP connections, maybe there is an error we are overlooking, a missing retry?

Best regard
Terje Nilsen
9Tek AS
Norway

gab933, 2022-03-08:
Thank you.
I'm attempting to solving the issue but I'm not able to understand which callback/service manages the internet/wifi connection.
Anyway I was able to manage the drone disconnection while app is running. In this way, it is possible to disconnect and reconnect drone (e.g., for changing battery) without restarting the app. If you are agree I can make a pull request on it.
Best Regards,

Gabriele

kripper, 2023-04-08:
Please check latest version.

closed 2023-04-08 by kripper