RosettaDrone/rosettadrone #109: Waypoint Action like START_TAKE_PHOTO not working

himanshu131098, 2022-05-20:
When developing missions with Waypoint Actions like START_TAKE_PHOTO, START_RECORD, STOP_RECORD I have observed that this works only with the last waypoint. The actions are not executed in the intermediate waypoints.

How can we add different waypoint actions? Has anyone developed a solution for this?

Thank You

Himanshu V
Vyorius Drones
New Delhi, India

The1only, 2022-05-23:
A new version is out, it introduces a loot of new things, so I branched the 1.14.x that we have used for the last year, just in case.
The new 1.16.1 should handle mission and waypoints better. 
Stuff like on-the-fly donwnload and mail repost og images is now supported, and work while still flying.

Phillweston, 2022-06-03:
I noticed that the last time the commit updated the live FTP image download function, but at present, the use of this function is not written in both README and wiki. So let me ask, if I want to take a photo in a waypoint mission, and export the photo to the QGroundControl end after the waypoint mission, how should I do it? Thank you

Phillweston, 2022-06-08:
# When using waypoint actions like taking photos in waypoint, an error occurs after uploading the mission plan:

## APP
[image]

## QGC
[image]

kripper, 2023-02-17:
Can be easily implemented in new MissionManager class. Do you still need it?
What exactly do you want?

a) Desired elapsed time between two consecutive pictures (in seconds). Minimum values depend on hardware (typically greater than 2 seconds)?
b) A total number of images to capture?

kripper, 2023-04-08:
Closing for now.

closed 2023-04-08 by kripper